hd-zero/hdzero-goggle #209: Feature Request: Add support for ESP NOW for integration with RotorHazard

Gierzow, 2023-04-08:
As it was recently added to TBS fusion module, since HDzero goggle already has esp32 it would be cool to recieve timer messages and print them in osd. As HD zero is aimed to change racing this feature should be one of top priorities.  Possition, current and last lap time and also lap number should be available to print. Would also be nice to be able to choose where those OSD elements displayed.

R5fan, 2023-10-02:
+1
And ideally you could receive a lap timer signal from your ELRS radio over backpack too. That way you can time laps manually using a momentary switch, or get some signal in to the radio via trainer port for a DIY solution. That would need support in elrs I imagine, but apparently something like that already exists?

Master92, 2023-11-11:
There is a plugin for Rotor hazard that seems to do exactly what you requested. Check it out here: https://github.com/i-am-grub/VRxC_ELRS

I think, this issue can in turn be closed.

Combinacijus, 2024-09-29:
Duplicate https://github.com/hd-zero/hdzero-goggle/issues/203