hd-zero/hdzero-goggle #223: Feature suggestion - standby mode

robhaswell, 2023-04-18:
Even with the new update, the boot time is very high. This can lead to you holding up a race if you forget to turn them on.

It might be possible to reduce power consumption to the point where the units don't need to be powered down. I suggest that a mode could be activated (by way of some button press, perhaps both buttons at once) to enter "standby mode".

In standby mode, the HDZero radios and OLED as powered down, and there could be additional firmware actions which reduce power consumption of the MCU. Fans are disabled or run at minimum. Standby mode ends as soon as any button is pressed.

Standby mode could also be entered automatically after a period of no signal, similar to the current OLED-off function.

pitts-mo, 2023-06-05:
@robhaswell a Go Sleep! function recently added and I have submit PR #268 with some improvements.

In your view of Standby mode... should the goggle attempt to immediately return to Live view on resume? or would you prefer it to return to the menu with "Go Sleep!" still highlighted?

Anyone else wish to weight in?
Thank you,
-p

pitts-mo, 2023-06-20:
As of 9.1.0-rc2 the Go Sleep! function appears to provide the basic hardware sleep functionality needed to resolve this issue. However, performance and usability are lacking.

Outstanding items I see are:
-make Go Sleep! function independent of menu.
-add logic for corded long left + right button press to actuate Go Sleep! function.

Sound correct?
If so maybe @robhaswell can update the title for this issue in a few days?

SAMUD, 2023-07-02:
I updated my googles to the lastet version and the GoSleep is cool. However some quick acess for this is needed.
Either both buttons or some kind of double click on the rec-button to send them to sleep.

Optional I think something like one beep every 30sec like on Orqa googles would be cool, so you are reminded that the googles are still powered on.

PR #348 (2023-11-03): Add shortcut for entering sleep mode from video
This PR extracts sleep mode code from the UI page to its own compilation unit, allowing it to be entered from different parts of the code.

I also added the shortcut of double-clicking the right button in video mode (not DVR playback) to enter sleep mode directly. Exitting from it should always return to the previous state.
When in sleep mode, a beep is output every thirty(-ish) seconds to remind the user that the goggles is still powered on (as suggested in #223) - this feature is open for discussion and can easily be removed if unwanted.

Close [... cut]
closed 2023-11-03 by ligenxxxx