hd-zero/hdzero-goggle #230: Feature Request: add AnalogRX Power option "Auto"

pitts-mo, 2023-04-24:
Please add "Auto" setting option to "AnalogRX Power" within the power settings page of UI.

If "Auto" 
check/set analog module power is ON when:
-goggle is booting and analog is the current mode or WiFi / other analog module bay feature is enabled
-mode is changed to analog or WiFi / other analog module bay feature changes to enabled
else
check/set analog module power is OFF when:
-goggle is booting and analog is NOT the current mode and WiFi / other analog module bay feature is NOT enabled
-mode is changed to NOT analog and all WiFi / other analog module bay features change to NOT enabled

pitts-mo, 2023-04-24:
$ grep -r -i power_ana
mkapp/app/setting.ini:power_ana_rx = 1
src/ui/page_power.c:    ROW_POWER_ANA,
src/ui/page_power.c:static btn_group_t btn_group_power_ana;
src/ui/page_power.c:        create_btn_group_item(&btn_group_power_ana, cont, 2, "AnalogRX Power", "On", "Off", "", "", ROW_POWER_ANA);
src/ui/page_power.c:    btn_group_set_sel(&btn_group_power_ana, g_setting.power.power_ana);
src/ui/page_power.c:    case ROW_POWER_ANA:
src/ui/page_power.c:            btn_group_toggle_sel(&btn_group_power_ana);
src/ui/page_power.c:            g_setting.power.power_ana = btn_group_get_sel(&btn_group_power_ana);
src/ui/page_power.c:            ini_putl("power", "power_ana_rx", g_setting.power.power_ana, SETTING_INI);
src/ui/page_power.c:            DM5680_Power_AnalogModule(g_setting.power.power_ana);
src/core/main.c:    DM5680_Power_AnalogModule(g_setting.power.power_ana);
src/core/settings.c:        .power_ana = false,
src/core/settings.c:    g_setting.power.power_ana = ini_getl("power", "power_ana_rx", g_setting_defaults.power.power_ana, SETTING_INI);
src/core/settings.h:    bool power_ana;
src/driver/dm5680.h:void DM5680_Power_AnalogModule(uint8_t on);
src/driver/dm5680.c:void DM5680_Power_AnalogModule(uint8_t on) {

pitts-mo, 2023-04-24:
Doubt I will be able to figure this out myself in any timely fashion. Will dump notes here as I find time.
-p

L1cardo, 2023-05-13:
This function would be so cool!

pitts-mo, 2023-06-02:
Please note it may be helpful to consider issue #267 when working this issue.

I remember thinking "src/core/settings.h: bool power_ana;" should be changed to byte but I have not made any real progress here.

pitts-mo, 2023-07-02:
@ligenxxxx 
Can you recommend a way to peruse this?

maybe change src/core/settings.h:
 bool power_ana;
 int power_ana;

This method would require changing settings option handling to result in 0,1,2 and updated processes to conditionally pass the appropriate 0 or 1 to DM5680_Power_AnalogModule() on boot and source state changes to keep analog module powered when analog is the currently selected source (even in the menu) .

pitts-mo, 2023-07-08:
Please thoroughly test PR #312 from SD card. -p

pitts-mo, 2023-07-13:
[HDZGOGGLE-PR312-2023-07-08.zip](https://github.com/hd-zero/hdzero-goggle/files/12044158/HDZGOGGLE-PR312-2023-07-08.zip)

pitts-mo, 2023-08-10:
Please test the new AnalogRX Power Auto found in 9.2.0-rc1.
https://github.com/hd-zero/hdzero-goggle/releases

pfeerick, 2023-11-04:
#312 has been merged, so this issue can be closed now?

pitts-mo, 2023-11-04:
Yeah, I guess this can be closed.

I had intended to come back and clean a few things after I originally submit PR #312... But I have not had time and most of the related ideas were in regard to audible notification handling and status displayed in the menu... all of these would better be served under their own distinct issues.

Thanks for the bump,
-p

closed 2023-11-04 by pitts-mo