hd-zero/hdzero-goggle #231: Feature Request: add stereoscopic 3D support

pitts-mo, 2023-04-24:
The HDZero goggle appears to be physically capable of displaying several stereoscopic 3D modes.

Suggest starting with SideBySide mode from HDMI input.

pitts-mo, 2023-04-24:
Can the HDZero receivers be split in order to display one RF channel on each screen?

lebed2045, 2023-07-11:
wow, that would be sick! also good for the company, so you can sell two vtx for the drone.

wernight, 2024-03-18:
Given how much "simple" VR headset make a buzz, having HDZero Google work as VR headset would actually up the value. They may not need to be great at it, just... good enough.

WooD1k, 2024-04-02:
It would be great to have a feature like this!

ewindisch, 2024-04-28:
This would be exciting for multiple use-cases. 3D FPV will soon be possible from a single VTX and camera using AI estimation with higher latency and/or reduced framerate, reducing weight and cost to these configurations.

ligenxxxx, 2024-06-07:
This will not be possible due to hardware limitations.

closed 2024-06-07 by ligenxxxx

konanast, 2025-07-25:
I was wondering if there is any information available regarding potential support for stereoscopic 3D in the new HDZero 2 goggles?

Is this feature being considered, or there is still technical limitation that would prevent it?