hd-zero/hdzero-goggle #235: Analog bay switching problems.

HazzaHFPV, 2023-05-01:
Running Firmware 13042023 (9.0.12) from HDZero Website.

Switching to analog via the menu does not work properly just gives me a black screen. Selecting "last" as input source and then rebooting the goggles the input from the analog module worked. If I then switch the source again to HDZero and back to analog: again black image. If I switch between NTSC and PAL, also black image. Only removing the analog module and plugging it in again or rebooting the goggles with analog selected with option "last" fixes it.

I experienced stuttering while flying. Switching between PAL and NTSC removed the stutter in the end..
I have a PAL camera but had to switch to NTSC in the goggles to remove the stutter, which is strange.

I also experienced rolling of the image at close proximity to myself with my rapidfire. Here I cannot confirm if the cause is to be found in the goggle firmware or somewhere else. It was just suprising as it happened only 30 meters away from me with vtx at 200mW.

gonebdg, 2023-05-06:
working fine with my tbs fusion.

oldrootbeer, 2023-05-12:
Any way of detecting the stutter in the software and auto switching between pal and ntsc?

pitts-mo, 2023-08-01:
Hello @HazzaHFPV & @lesharodin  , Has this issue been resolved?
If not please attempt a full five file Goggle update from Rev_26062023.

AUTO might be a nice feature for brief testing. But most people use FPV systems for remote operations where NTSC or PAL signal processing can not be trusted to "AUTO" anywhere along the signal path.

lesharodin, 2023-08-01:
For me it was resolved when Carl fixed it first time :)

Dihckno, 2024-03-30:
I have the same issue woth tbs fusion, and the menu of tbs fusion  mirror on osd is affcted too, need to change manually ntsc and pal and return the good signal.   Version 9.31 rx  75 va 191.