hd-zero/hdzero-goggle #244: DVR Not Recording

hornedreaper33, 2023-05-10:
Hey team. I am on the latest firmware app: 9.0.12 rx: 72 va: 161, and my DVR is not recording again. Record is flashing in the OSD, 128Gb card formatted in the goggles. It will record down to ~108 Gb remaining and there it stays.

pwojt, 2023-05-13:
Running into the same issue. I was ablee to format the card and get DVR to record, but after reinserting it no longer records. No idea why since I'm still seeing the REC indicator.

gonebdg, 2023-05-23:
> Hey team. I am on the latest firmware app: 9.0.12 rx: 72 va: 161, and my DVR is not recording again. Record is flashing in the OSD, 128Gb card formatted in the goggles. It will record down to ~108 Gb remaining and there it stays.

is this issue still happening if you use <= 32Gb card ?

pwojt, 2023-05-23:
I had this issue come up twice now on a 64gb card. Both time it was fixed by formatting the card, but I’m not able to figure out how it got in that state.

I tried to reproduce this issue without any luck recently. Will add additional details if I’m able to repo.

gonebdg, 2023-05-24:
Lucky me never had this problem.
32Gb card, Record = Auto, format = .ts, osd embeed, power lipo 2s & 3s never goes to warning.

pwojt, 2023-05-24:
Interestingly I had this happen again today. Same settings as you, here is what I did that seemed to cause it:
* Flew HDZero quad, but got into some trouble with motors.
* Turned off goggles before powering down quad
* Turned everything back on and went flying with another HDZero quad
* Checked playback menu after and video was not recorded

Wonder if this is an issue with previous recording not closing properly? Once I opened playback and started flying again video was recording no problem. Seems like I'll need to review playback menu after flights to make sure DVR is recording. I would try a 32gb carb, but all of mine are slower so worried I will have other issues.

gonebdg, 2023-05-24:
That's weird. 
The video .ts format file supposed to prevent recording problem that usually happen with .mp4 format when VTX accidentally shutdown during crash or at low power voltage.
I just use my cheapest "sand disk" card class 10 instead of the A1 or Extreme
https://pasteboard.co/MKZzvY7Ke87B.png

daviddhall, 2023-05-26:
I posted this on discord: Discovered that none of my flying today recorded. I'm positive the "Recording" symbol was lit up red because I was trying to make an IGOW video and verified it at various points of flying today. I'm using a 128 GB card. It has video recordings on it from previous days. I had viewed the files on it before flying today where it gives that error "problem with this drive. Scan the drive....". I see that on >32GB drives formatted as FAT32. I wonder if that somehow prevented recording. SD card is not out of space.

daviddhall, 2023-05-27:
Today, I formatted the 128GB card in the goggles, and flew a few batteries with the same hdzero whoop. It didn't record the last several flights. In between the ones that worked and ones that didn't, I unplugged the google battery. (I might also be missing the flight before I unplugged, not sure - meaning I might have done it before it finished saving) 

I'm going to switch to a 32GB card and see if that matters any.

SAMUD, 2023-05-27:
Same problem here.
Auto record, *.ts selected.

Last week i had it three times where the goggles showed "rec" but on my sd card where only files from 0 to 8. All the other files where just not here. (forgot which size card I was using. I think 64GB)

After switching the card to another 64gb one the goggles still showed "rec", but now there aren't any files on the card. I tried formatting it in Windows and the goggles, but It didn't change anything.

Next try is with a 16Gb card I have lying around.

FlyingChad, 2023-05-27:
Same thing is happening to me. It seems to happen as soon as I pull files off the SD card in windows. As soon as I do that, everything seems like its working, REC on OSD. But when you look on the SD there are no files. Format card in the goggles and it starts recording properly again.

gonebdg, 2023-05-28:
Maybe worth to mention that I still use the original goggles firmware 9.0.1 rx: 72 va: 150 
[SanDisk 32Gb card class 10 ](https://pasteboard.co/MKZzvY7Ke87B.png)  (the cheapest one, not A1 nor extreme) 
Record = Auto, format = .ts
no problem wirh recording (hdz & analog)

SAMUD, 2023-06-01:
It seems, that with the 16gb card, formatted inside the googles the recording is working again.
Files are there, and already 15clips on it.

SumolX, 2023-06-03:
Try this PR https://github.com/hd-zero/hdzero-goggle/pull/269

SAMUD, 2023-06-25:
Recording with a 16GB card worked for me without any problems until yesterday.
Yesterday, Rec was showing up, but since file 044 the goggles just didn't record anything anymore.

SumolX, 2023-06-25:
> 

Which firmware version were you running?  Please contact me directly on discord via SX

SQuiDFPV, 2023-06-26:
23 packs, race qualfying and not one single dvr recorded, the REC was flashing on everyone.
i had tested the day before, video DVR worked. went and flew the quals today, (need DVR to qualify) nothing recorded.
latest FW Auto,Mp4 (i unluggged quad, set the next quad up and plugged in, rec shown in OSD but still no recordings. no qualifying for me this year :(

closed 2024-06-07 by ligenxxxx