hd-zero/hdzero-goggle #250: Feature Request: Instructions on Focus Chart page

evilC, 2023-05-20:
When loaning your goggles to someone to try them out, you would generally put the goggles onto the Focus Chart page, as that is the first thing they will need to do.  
Why not put some instructions on this page? For example, some text pointing to where the focus/IPD knobs are (And how to use them), plus some text pointing to the top left saying how to exit out of the menu (eg "Long press button HERE twice to exit to FPV view")

AlexandreBonneau, 2023-06-03:
[image]
Adding instructions to change the IPD is hard on the existing focus chart, since there is not much space left.
Also, if you lend your goggles to somebody, you can just explain that to them.

I've added some filler text all around the focus chart because while having the circles is handy,I find it important to get text that you ought to be able to read everywhere on the screen as well.

evilC, 2023-06-03:
> Adding instructions to change the IPD is hard on the existing focus chart, since there is not much space left. Also, if you lend your goggles to somebody, you can just explain that to them.

Agreed, not that big of a deal as they are in a pretty standard place.  
Looks good!

ligenxxxx, 2024-06-07:
TODO