hd-zero/hdzero-goggle #251: DVR get deleted on click, not hold.

lesharodin, 2023-05-21:
In playback menu it says hold right button to delete video file, but it gets deleted even on short press.

pitts-mo, 2023-05-23:
Confirmed: 
I unfortunately discovered this today while testing #249.

Not only should it be a hold press it should present a warning dialog to confirm delete with a quick enter (center wheel) press.

lesharodin, 2023-06-21:
Fixed in new fw

pitts-mo, 2023-06-21:
@lesharodin I find this is still occurring for me on 9.1.0-rc3 and think this should be keep open until a warning dialog is added. 

Can you double check a quick click of left button is not deleting the selected DVR, verify the firmware version installed on your goggle, and that you are not running an old build from SD card?

lesharodin, 2023-06-21:
@pitts-mo Yeah, sorry.
For some reason only hot files not get deleted by click, other files still gets deleted, my approach in PR to fix it by increasing time of button pressed seems not helping.
Reopened

lesharodin, 2023-06-21:
For example in Wifi Keyboard menu Long press and  click of right button works well

wxyzsupermod, 2024-09-23:
@lesharodin I added this after after deleting my DVR on accident trying to hot it. Addressed in https://github.com/hd-zero/hdzero-goggle/pull/443