hd-zero/hdzero-goggle #271: After video files hdz_1000.* next file number reverts to hdz_101.* causing file overwrite issues in automation

RoGoR1, 2023-06-04:
I need to maintain unique file names during offloading of video files, and I am trying to maintain a time based name where at least the name represents an order in time since time is not reliably maintained.

After file hdz_1000.ts/jpg the next file number becomes hdz_101.ts. This screws up video file SD card offloading automation.The automation process I have always used is to leave the the last file on the SD card in order to maintain the numbering for the next recording and maintaining unique file name to all destination file names. This file naming method breaks down after file 1000. The last ~50 files automation has been overriding earlier video file ~hdz_101.ts - hdz_150.ts in the target directory.

evilC, 2023-06-08:
I am not sure this is really feasible - the names would keep getting longer and longer. After a while, the numbers would simply not fit in the available space for each video in the playback menu.
Maybe a better solution in your case would be some kind of program / script on your PC which copies the files for you and does the re-number?

RoGoR1, 2023-06-08:
I see your point, but it still sounds like a bug not a feature. ;) I've
got the script for mounted sd card figured out. Once I start leveraging
rsync over wifi, it's going to become more complicated. I suppose I'll
just make a holding directory. This logic tries to keeping the
numbering, but in the end, it's already built in to deal with numbering
not being preserved. So a big chunk of this code could go. It allows
for destination files that have had the name modified. I always
preserve hdz_nnn* but often tag on a descrtipion
hdz_1134_cops_trying_to_pull_over_drone.ts. A big chunk of this code
could go. Yes, it accomplishes the task. But I doubt I'm the only one
in this boat. Dates are bigger numbers, will we have that for names?
---------------------------------------------------------------------

#!/bin/bash

target='/sdcard/mount'
destination='/destination/movies'
script='/destination/offload_movies.script'

first_source_file_number=$(ls -1 ${target}/hdz_[0-9]*.*|cut -d'_' -f2 | cut -d'.' -f1 | sort -n |head -1)
last_source_file_number=$(ls -1 ${target}/hdz_[0-9]*.*|cut -d'_' -f2 | cut -d'.' -f1 | sort -n |tail -1)
last_destination_file_number=$(ls -1 ${destination}/hdz_[0-9]*.*|cut -d'_' -f2 | cut -d'.' -f1 | sort -n |tail -1)
let next_destination_file_number=${last_destination_file_number}+1

# File may or may not have numbers. Probably wont be 237 matching 2370 because then we would be working
# with 2370. Still safer to nail down the expression. In Bash *([^0-9]) is 0 or more non digits, but
# follow with a * to match the rest of the string that could be anything including numbers.
shopt -s extglob
last_destination_file=$(echo ${destination}/hdz_${last_destination_file_number}*([^0-9])*.ts)

echo "INFO: First source file number: ${first_source_file_number}"
echo "INFO: Last source file number: ${last_source_file_number}"
echo "INFO: Last destination file number: ${last_destination_file_number}"
echo "INFO: Last destination file: ${last_destination_file}"

# If there is only one file on the card and it already matches last destination, we are done
# If the source matches the destination AND there are more files to process, delete the source and incriment the first_source_file_number
if [ "$(md5sum ${target}/hdz_${first_source_file_number}.ts ${last_destination_file}| awk '{print $1}' | uniq -d)" != '' ]; then
 echo "INFO: The following files are equal in content:"
 echo "      ${target}/hdz_${first_source_file_number}.ts"
 echo "      ${last_destination_file}"
 if [ ${first_source_file_number} -eq ${last_source_file_number} ]; then
  echo "INFO: There is nothing to do here leaving ${target}/hdz_${first_source_file_number}.* on ${target}"
  exit 0
 fi
 echo "INFO: new first source file is: ${target}/hdz_${first_source_file_number}.ts"
 echo "INFO: Removing source file that already exists on the target."
 echo ""
 echo "rm -f ${target}/hdz_${first_source_file_number}.*" | tee ${script}
 let first_source_file_number+=1
fi

# Copy over source to mapped destination.
for i in $(seq ${first_source_file_number} ${last_source_file_number}); do
 # Skip missing or deleted video files
 if [ -f ${target}/hdz_${first_source_file_number}.ts ]; then
  if [ ${first_source_file_number} -eq ${last_source_file_number} ]; then
   echo "cp ${target}/hdz_${first_source_file_number}.ts ${destination}/hdz_${next_destination_file_number}.ts" | tee -a ${script}
  else
   echo "mv ${target}/hdz_${first_source_file_number}.ts ${destination}/hdz_${next_destination_file_number}.ts" | tee -a ${script}
   echo "rm -f ${target}/hdz_${first_source_file_number}.jpg" | tee -a ${script}
  fi
  let next_destination_file_number+=1
  let first_source_file_number+=1
 fi
done

echo "DONE!" | tee -a ${script}
ls -ltr ${target} | tee -a ${script}

echo "INFO: This was a dry run. cat/bash ${script}"
exit 0

---------------------------------------------------------------------------

evilC, 2023-06-09:
Is the optimal solution not to just do the RTC mod? That will give you order out of the box.

RoGoR1, 2023-06-09:
> Is the optimal solution not to just do the RTC mod? That will give
> you order out of the box.

Kind of, yes, maybe, soon.

I did in fact create and install a RTC battery. While I am now getting
relative timestamps I am also get an error while setting the time, and
the naming did not change. So either I did something wrong or there is
an issue with that too. But I figure if there is an issue someone else
is going to catch it.

The numbering system has grown on me, I kind of like it. It's what we
are presented with out of the box and what I am going with for now. Not
everyone is going to want to open up their goggles and install a
battery. So while I am eccentric, there are likely others that have the
same idea given what they are presented with. They are going to hit the
same bug (yes it's a bug). Hopefully they won't blindly overwrite a 100
or so files while not paying attention. Obviously I had some faults of
my own.

This is not critical, but if I where developing this code, I defiantly
want to know about the bad logic (jumping from 1000 to 100 is certainly
without intent). That's me though, I don't like faulty logic even if I
can't necessarily think of a reasonable conditions where the exception
is going to be met.

- Roger

evilC, 2023-06-09:
I don't think the naming changes (I only just installed mine, so dunno), it's the file creation date of the file that would tell you the date+time

ligenxxxx, 2024-06-07:
You can use rtc and name it with time to solve this problem.

closed 2024-06-07 by ligenxxxx