hd-zero/hdzero-goggle #289: automatic recording does not work well with analogue

Seba112PL, 2023-06-19:
automatic recording does not work well with analogue. It records all the time even noise

pitts-mo, 2023-06-20:
I have never liked auto record function on any goggle. I use manual record and prefer it records everything because sometimes a record of no signal/poor signal duration is very helpful (reference ending of "Contact" 1997)... So, It seems more alighted with the intent of DVR (Auto or manual ) to record everything by default. 

However, it would be nice to have a separate sub option for recording to file that allows "always" or "only when signal is locked" so the user can choose from the best of both worlds.

In case this helps you now:
I use ELRS and set my TX with backpack (LUA) to start goggle recording on ARM and end goggle recording 15 seconds after disarm. This works for any input the goggle can record and even plays well with my preference to manually start recording before I even plug in my aircraft (to capture FC splash screen and version info).

thespova, 2023-09-23:
This has not been working for me either since updating from 8 to 9 on 1st gen goggles.  Just tried updating to the latest version 9.2 and its still not working.

R5fan, 2023-10-02:
If you use elrs, and have the wifi module, you can use the backpack feature to start recording upon arming and until x seconds after disarming (find those settings in the elrs lua script). Works perfectly and its the only use i have found for backpack so far :)