hd-zero/hdzero-goggle #295: Emulator not working on windows / devcontainer

evilC, 2023-06-22:
Have WSL2 and Docker Desktop installed

Originally I also could not get the firmware build to work either, but it seems selecting this option in VSCode fixes that:
[image]
(Something to do with symlinks apparently)

I seem to be able to build the emulator just fine
```
vscode ➜ /workspaces/hdzero-goggle/build_emu (main) $ cmake .. -DEMULATOR_BUILD=ON -DCMAKE_BUILD_TYPE=Debug
-- Configuring done
-- Generating done
-- Build files have been written to: /workspaces/hdzero-goggle/build_emu
vscode ➜ /workspaces/hdzero-goggle/build_emu (main) $ make -j `nproc`
[...]
[100%] Linking C executable HDZGOGGLE
[100%] Built target HDZGOGGLE
```

However, when I try to run the emulator executable, I get no window
```
vscode ➜ /workspaces/hdzero-goggle/build_emu (main) $ ./HDZGOGGLE 
touch: cannot touch '/mnt/app/setting.ini': No such file or directory
[INFO][rtc.c:rtc_init:194] rtc_init has NOT detected a battery
[INFO][rtc.c:rtc_init:198] rtc_init updating both clocks via settings
[ERROR][i2c.c:iic_is_port_ready:39] Device 1:/dev/i2c-1 is not available [0=N/A,1=Right,2=Main,3=Left]
[ERROR][i2c.c:iic_is_port_ready:39] Device 2:/dev/i2c-2 is not available [0=N/A,1=Right,2=Main,3=Left]
[ERROR][i2c.c:iic_is_port_ready:39] Device 3:/dev/i2c-3 is not available [0=N/A,1=Right,2=Main,3=Left]
[INFO][mcp3021.c:aduc_read_voltage:60] Voltage ADC is mcp
[ERROR][mcp3021.c:mcp_read_voltage:31] mcp_read_voltage: open error.
[INFO][common.c:bmi2_interface_init:95] I2C Interface 
[ERROR][common.c:bmi2_error_codes_print_result:159] Error [-2] : Communication failure error. It occurs due to read/write operation failure and also due to power failure during communication
[INFO][accel_gyro.c:init_bmi270:74] init_bmi270 done.
[ERROR][common.c:bmi2_error_codes_print_result:159] Error [-2] : Communication failure error. It occurs due to read/write operation failure and also due to power failure during communication
[ERROR][common.c:bmi2_error_codes_print_result:159] Error [-2] : Communication failure error. It occurs due to read/write operation failure and also due to power failure during communication
[ERROR][common.c:bmi2_error_codes_print_result:159] Error [-2] : Communication failure error. It occurs due to read/write operation failure and also due to power failure during communication
[ERROR][accel_gyro.c:enable_bmi270:91] [Error] BMI270 failed.
[INFO][fans.c:fans_top_setspeed:24] fans_top_setspeed: 4
error: XDG_RUNTIME_DIR not set in the environment.
error: XDG_RUNTIME_DIR not set in the environment.
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Scan Now
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Source
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Image Setting
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry OSD
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Power
[INFO][page_power.c:page_power_update_cell_count:50] cell_count:2
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Fans
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Record Option
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Auto Scan
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry ELRS
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry WiFi Module
sh: 1: /mnt/app/script/wlan_stop.sh: not found
Changing password for vscode.
Current password: passwd: Authentication token manipulation error
passwd: password unchanged
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Head Tracker
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Playback
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Storage
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Firmware
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Focus Chart
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Clock
[DEBUG][ui_main_menu.c:main_menu_create_entry:254] creating main menu entry Go Sleep!
[WARN][oled.c:OLED_Startup:246] OLED_Startup failed: Auto init is not ready...
[INFO][oled.c:OLED_Startup:269] OLED temp test: 0xD0 L = 0  0  0  0  0  
[INFO][oled.c:OLED_Startup:270] OLED temp test: 0xD0 R = 0  0  0  0  0  
[INFO][oled.c:OLED_Startup:300] OLED temp test modified: 0xD0 L = 0  0  0  0  0  
[INFO][oled.c:OLED_Startup:301] OLED temp test modified: 0xD0 R = 0  0  0  0  0  
sh: 1: dispw: not found
sh: 1: aww: not found
[INFO][oled.c:OLED_display:504] OLED: Display off
[INFO][oled.c:OLED_display:497] OLED: Display on
[INFO][osd.c:load_fc_osd_font_bmp:729] load_fc_osd_font_bmp: /mnt/extsd/resource/OSD/FC/BTFL_000.bmp...
[ERROR][osd.c:load_fc_osd_font:835]  failed!
[INFO][osd.c:load_fc_osd_font_bmp:729] load_fc_osd_font_bmp: /mnt/app/resource/OSD/FC/BTFL_000.bmp...
[ERROR][osd.c:load_fc_osd_font:835]  failed!
[INFO][osd.c:load_fc_osd_font_bmp:729] load_fc_osd_font_bmp: /mnt/app/resource/OSD/FC/BTFL_000.bmp...
[ERROR][osd.c:load_fc_osd_font:835]  failed!
[INFO][osd.c:load_fc_osd_font_bmp:729] load_fc_osd_font_bmp: /mnt/extsd/resource/OSD/FC/BTFL_FHD_000.bmp...
[ERROR][osd.c:load_fc_osd_font:835]  failed!
[INFO][osd.c:load_fc_osd_font_bmp:729] load_fc_osd_font_bmp: /mnt/app/resource/OSD/FC/BTFL_FHD_000.bmp...
[ERROR][osd.c:load_fc_osd_font:835]  failed!
[INFO][osd.c:load_fc_osd_font_bmp:729] load_fc_osd_font_bmp: /mnt/app/resource/OSD/FC/BTFL_FHD_000.bmp...
[ERROR][osd.c:load_fc_osd_font:835]  failed!
[INFO][ui_main_menu.c:menu_reinit:222] menu_reinit
[INFO][fans.c:fans_right_setspeed:38] fans_right_setspeed: 2
[INFO][fans.c:fans_left_setspeed:32] fans_left_setspeed: 2
[INFO][fans.c:fans_right_setspeed:38] fans_right_setspeed: 2
[INFO][fans.c:fans_left_setspeed:32] fans_left_setspeed: 2
[INFO][TP2825.c:TP2825_Set_Clamp:257] Clamp = 7f
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][nct75.c:nct_read_temperature:36] nct_read_temperature: [Top] side error.
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[INFO][dm6302.c:DM6302_init:1603] Init0 done
[ERROR][dm6302.c:DM6302_init:1613] Error: DM6302s have no response.
[INFO][hardware.c:HDZero_open:207] HDZero: open
[INFO][page_scannow.c:scan_channel:277] Scan channel0: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel1: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel2: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel3: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel4: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel5: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel6: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel7: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel8: valid:0, gain:61
[INFO][page_scannow.c:scan_channel:277] Scan channel9: valid:0, gain:61
[INFO][page_scannow.c:page_scannow_enter:364] scan return :-1
[INFO][ui_main_menu.c:submenu_exit:167] submenu_exit
[INFO][hardware.c:HDZero_Close:216] HDZero: close
```

I notice: `error: XDG_RUNTIME_DIR not set in the environment.`  
Was asking around in Discord, and someone seemed to think that maybe X was just fundamentally broken.  
He suggested trying an X application to see if it worked 
```
vscode ➜ /workspaces/hdzero-goggle/build_emu (main) $ xeyes
bash: xeyes: command not found
vscode ➜ /workspaces/hdzero-goggle/build_emu (main) $ sudo apt-get install x11-apps
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libxaw7 libxft2 libxkbfile1 libxmu6 libxmuu1 libxpm4 xbitmaps
Suggested packages:
  mesa-utils
The following NEW packages will be installed:
  libxaw7 libxft2 libxkbfile1 libxmu6 libxmuu1 libxpm4 x11-apps xbitmaps
0 upgraded, 8 newly installed, 0 to remove and 6 not upgraded.
Need to get 1,144 kB of archives.
After this operation, 3,840 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://archive.ubuntu.com/ubuntu jammy/main amd64 libxmuu1 amd64 2:1.1.3-3 [10.2 kB]
[...]
Processing triggers for man-db (2.10.2-1) ...
vscode ➜ /workspaces/hdzero-goggle/build_emu (main) $ xeyes
Error: Can't open display: 
```

evilC, 2023-06-22:
Edit: Starting to realize that I may be being a bit dumb...
Do I maybe need something running on my windows host that can render the X window?

lesharodin, 2023-06-23:
Same, I couldt even build a fw with docker devcontainer on WIndows.
Ended up using WSL, it just a straight forward works, both building fw and running emulator.