hd-zero/hdzero-goggle #298: Garish image settings since RC1

MartinHugh, 2023-06-25:
Sometime in between RC1 and RC3, the image settings appear to have been changed.

The greens and the reds are significantly intense, so I assume that the underlying saturation settings have been modified.
Note that even moving the goggle image saturation settings down into single figures does not have a huge affect.

If this was to fix something in analog, that is fine, but note that it also affects digital - which were, just fine (for me at least)

I dont know if someone will notice this and dumb the settings down in the source for the next release when they realise how bad it is.
However meanwhile I have 10 or so digital aircraft I am going to have to go around and adjust the image settings for.
Unless the changes are going to be backed out.  I really dont want to adjust everything twice :-) 

Some communication would be useful in the release note on the rationale for this change and advice to users what the consequences of this "fix" would be and what they should do to accommodate it.
Or ideally, put it back as it was, as it was just fine for me.

Thanks for reading

SumolX, 2023-06-26:
The rational for this change was unify the behavior between "Reset All Settings" within Firmware Page and Image Settings Page.  When I updated the code, I based it off the Image Settings page "reset image settings" hardcoded defaults which the values were wrongly coded when compared to a setting.ini file that has not been modified.  Ryan proposed what values should be restored and after much youtube scraping we were able to restore all values.

Values are as follow thus far for defaults/reset all settings:

OLED: 8
Brigthness: 39
Saturation: 28 (Used to be 25 but Ryan suggested 28 would look better)
Contrast: 25
OLED Auto Off: 5 min

MartinHugh, 2023-06-26:
Thank you for the information, I assume these values will stay fixed now, and I will make adjustments to each aircraft for to accommodate the changes.

closed 2023-06-26 by MartinHugh