hd-zero/hdzero-goggle #304: Rename the firmware zip files so that they get ordered alphabetically, and thus chronologically

AlexandreBonneau, 2023-07-01:
Currently, the firmware zip files are named in the format `Rev_ddmmyyyy`.
This is wrong since by default the file system will order the files alphabetically, which means the firmware zip files are _not_ ordered chronologically, making it harder each time to identify which version is the last.

A simple way to fix this is to rename the generated zip files using the ISO date like `yyyy-mm-dd`.
So instead of having (where the chronological order is written in parenthesis) :
```
Rev_04042023.zip (5)
Rev_06012023.zip (3)
Rev_06022023.zip (4)
Rev_11062022.zip (1)
Rev_13042023.zip (6)
Rev_22122022.zip (2)
Rev_26062023.zip (7)
```

You'd get the much more clean and useful firmware list:
```
2022-06-11_HDZero_Firmware.zip (1)
2022-12-22_HDZero_Firmware.zip (2)
2023-01-06_HDZero_Firmware.zip (3)
2023-02-06_HDZero_Firmware.zip (4)
2023-04-04_HDZero_Firmware.zip (5)
2023-04-13_HDZero_Firmware.zip (6)
2023-06-26_HDZero_Firmware.zip (7)
```

Also for clarity, it could better to actually name what the zip file is, i.e. the HDZero firmware.

For Linux users, you can rename your existing badly named firmware files using this string in Krename:
`[$9-[length]]-[$7;2]-[$5;2]_HDZero_Firmware`.

matyas-brenner, 2023-09-14:
Relates to https://github.com/hd-zero/hdzero-vtx/issues/154

ligenxxxx, 2024-06-07:
sloved.

closed 2024-06-07 by ligenxxxx