hd-zero/hdzero-goggle #307: [Feature Request] Add ext4 and exFat file system support

antony-weber, 2023-07-03:
Add ext4 and exFat sd card file system support

christhomas, 2023-07-24:
Apparently the Tina Linux source tree doesn't contain the code for these file systems. But I think that's just a case of adding it, compiling it, and trying ti out. Just adding this information here because I was told it in a different issue

antony-weber, 2023-07-24:
Ext4 exist in Tina Linux source tree

christhomas, 2023-08-18:
Ah, yeah sorry, you’re right, I should have only mentioned exfat

zandorsp, 2023-08-29:
I think F2FS filesystem [https://en.wikipedia.org/wiki/F2FS](url) is a better option, because it's optimized for SD, transparent compression and atomic operations.

christhomas, 2023-09-01:
mountable in windows and macos without any drivers to be installed?

closed 2024-06-07 by ligenxxxx