hd-zero/hdzero-goggle #310: Turn on the required display while focusing, turn off the other display

MARUTHI344, 2023-07-07:
Add option to turn on the required display while focusing, keep the other display off.
Sub menu 
Both displays
Right display
Left display

oldrootbeer, 2023-07-07:
Nice idea, But isn't it sufficient to just close one eye?

MARUTHI344, 2023-07-07:
> Nice idea, But isn't it sufficient to just close one eye?

This request has come from my personal experience, I tried focusing using the regular way, that is closing one eye. The results were not satisfactory. Checked with the discord community, based on the input, I could understand that keeping the eye relaxed while focusing was key. Closed my eyes using a eye patch, one at a time. Results were better. Even the ophthalmologist uses a blank to blind one eye and test the other eye to get the right power. So, the closet simulation would be to turn off the screen(non focusing). The goggle foam pad does a good job in preventing the light leak from outside.

Mod-Quad, 2023-08-24:
This is a fantastic idea as most people cannot independently close their dominant eye.  And closing one eye tends to affect the focus of the open eye.  I urge the devs to incorporate this proposed functionality.