hd-zero/hdzero-goggle #311: [feature request]AI corrected signal loss.

Shaun-York, 2023-07-07:
Would it be possible to train a computer vision model with OpenCV to fill in lost image data in realtime?

nurikk, 2023-07-11:
And AI 4k upscale 😍

robhaswell, 2023-07-11:
You would need to add a GPU into the video path. I'm certain this would be a fun research project but there is no chance of seeing this come included with HDZero.

lebed2045, 2023-07-11:
jokes aside, it seems possible to pre-train model to predict the most statistically optimal pixel color based on neighbours so actual pixel filaments can be done on fly with very low computation overhead. Another option, but I'm sure developers explored it already, is to dedicate small part of bandwidth to send colors of big areas first, i.e. like old JPEGs use progressive loading in the past.

xtr0d3m0n, 2023-07-25:
Depending on how much hardware u got on v536. It got an NPU that can probably do some decent inference.

francoisbeaussier, 2023-08-27:
Would  averaging the neighbouring pixels already be an improvement over the current "multicolour static" behaviour?

ligenxxxx, 2024-06-07:
Any software-based video processing will increase latency.

closed 2024-06-07 by ligenxxxx