hd-zero/hdzero-goggle #313: Better automatic analog/expansion module power toggling when selecting expansion module as a source

Combinacijus, 2023-07-12:
## Problem
When flying HDZero and analog quads to swap between the two it's needed to:
1. Scroll to Power tab
2. Scroll to Expansion Module power setting
3. Enable Expansion Module power
4. Go back to Source tab
5. Select Expansion Module as a source
6. After flying Scroll to Power tab
7. Disable Expansion Module power to save battery

That's a lot of steps

# Solution
1. Automatically turn on Expansion Module power when Expansion Module is selected as a source and turn off power when back in menu
2. In a case that's not the desired behavior for some pilots Power tab could have auto option: [>Auto] [On] [Off]
3. [Debatable if needed] To reduce analog module boot time on Auto mode analog module could be powered whenever Source tab is opened. 
   - Possible problems with this would be: scrolling through the tab would trigger power on; on boot analog module might beep which could get annoying.
   - Giving an option to enable might be a fair solution: [>Auto] [Auto Fast Boot] [On] [Off]

So now there would be way less steps:
1. Go to Source tab
2. Select Expansion Module as a source (power of the module is handled automatically)

pitts-mo, 2023-07-13:
Please test this HDZGOGGLE file from SD card.

This file was built from PR #312 which replaces AnalogRX Power OFF with a new "Auto" function. It has been coded to leave analog module on when it is the currently selected mode even while working within the menu. I have found changing the power state too frequently can cause problems with analog module restarting and re-aquiring signal quickly as well as loosing ELRS backpack binding.

[HDZGOGGLE-PR312-2023-07-08.zip](https://github.com/hd-zero/hdzero-goggle/files/12044158/HDZGOGGLE-PR312-2023-07-08.zip)

pitts-mo, 2023-08-10:
Please test the new AnalogRX Power Auto found in 9.2.0-rc1.
https://github.com/hd-zero/hdzero-goggle/releases

Combinacijus, 2023-08-10:
@pitts-mo Tested 9.2.0-rc1. Auto analog module power works well, tested with various cases I could think of. 
Thank you, it's great

pfeerick, 2023-11-04:
#312 implements this and has been merged, so this issue can be closed now?