hd-zero/hdzero-goggle #327: Enhance audible notifications -for discussion

pitts-mo, 2023-08-05:
Since I received my goggle I have wanted enhanced audible notifications and controls for audible goggle feedback. However, after PR #312 I suspect more people may have desires/suggestions for audible feedback handling/behavior . So I wanted to open this issue with some ideas for discussion.

I am currently thinking audio notifications should have:
1. priority values for all audio notifications.  Say 0-5(?) with highest number lower being lowest priory.
2. menu option/slider to select the desired priory level for audio feedback. From Critical(0) to All(5).
3. set zero priority by default to force all existing notifications to produce feedback until they are re-coded for a reduced priority level
4. add a volume control mechanism for notifications with priority >=1 
5. menu option/slider to select volume 0-10(?).

It may also be desirable to create different beep types or distinct beep types at different priorities?

pitts-mo, 2023-08-05:
Some examples of where I desire audible feedback from goggle:
-Boot complete (*&*)
-record start
-record stop
-wake from Go Sleep! (*)

maybe:
-enter sleep
-record failure?
-SD card failure?

* = PR312 incidentally adds this for V2 goggle users 
&* = only if AnalogRX Power is not needed after boot completes (v2 goggle)

evilC, 2023-09-04:
The following may benefit from audio notifications
#338 Beep when switching to FPV view is aborted
#337 Beep when long press action is triggered
In both cases, this would let the user know when the button can be released, as the desired action has taken place