hd-zero/hdzero-goggle #335: Low RF frequencies 2.4GHz 1.3GHz

Yury-MonZon, 2023-08-27:
Divimath mentioned here http://www.divimath.com/dm5680-en/

`22km (520MHz, 30dBm)` 

From this I suppose it is possible to use frequencies lower than 5.8Ghz ie 2.4GHz, 1.3GHz. 

I wanted to add the support for lower frequencies but I can't find datasheets for the DM6300 RF chip.

Anyone has an access to that information?

Master92, 2023-11-05:
Just a guess: Divimath's chip does indeed support it but I guess HDZero is explicitly intended for the 5.8GHz band (maybe due to bandwidth?)

If I remember correctly you have to contact Divimath directly for getting the chips and the reference manuals. Maybe mailing them could give you the desired answers.

ligenxxxx, 2024-06-06:
HDZero Goggle is designed for 5.8GHz, and even if it works on other channels, the performance will be very poor. We have no plans to support other frequencies at this time.

closed 2024-06-06 by ligenxxxx