hd-zero/hdzero-goggle #338: Feature request: Abort switching to FPV view on boot if button held

evilC, 2023-08-30:
If Auto-Scan is set to last, when it boots up, it shows the menu, but then once it loads it goes to FPV view.  
This is nice if you just want to fly without changing anything, but eg if you want to tweak settings or go to playback, then it's annoying.  
Can we not have it set up so that if you hold the action button while it boots, it stays on the menu and does not go to the FPV view?

cougman, 2023-08-31:
The microphone still doesn’t work on analog unless you start with hdzero and then switch
Why?

Sent from myMail for iOS

ligenxxxx, 2024-06-07:
set auto scan to off.

closed 2024-06-07 by ligenxxxx

evilC, 2024-06-07:
> set auto scan to off.

I generally want it on, but sometimes you just want to go straight into the menu on bootup