hd-zero/hdzero-goggle #340: Feature Request: Enable/Disable OSD Recording in goggles from TX

Yith1, 2023-09-04:
Ability to switch OSD Recording to files on goggles by a switch on the Open/Edge TX transmitter.

This would involve sending the signal through the ELRS Backpacks in a similar way to the VTX Channel change feature.

An important factor would be the ability to do this during flight. The idea being you can see the OSD on the screen whilst flying and be able to enable/disable the OSD recording when you go through an area which you want to record without OSD.

It would be useful to be able to see an indication on the display as to whether the feature is on/off.

An alternate option would be to have a mode where the OSD data is stored in a separate file from the Video data, similar to Walksnail.

pitts-mo, 2023-09-04:
I like this idea.

I turn Off OSD in Betaflight and Inav and was also thinking it would be nice to key goggle display state based on MSP display port state. So Goggle OSD would need three options: On / Off / Follow MSP.

-Does MSP display port data indicate when OSD has been set off?
-If not can we disable goggle OSD when MSP link data is detected and when MSP display port has no elements displayed excluding warning messages.

pitts-mo, 2026-09-12:
issue for ELRS backpack  ExpressLRS/Backpack/issues/246