hd-zero/hdzero-goggle #341: [Feature Request] Long press of Func button breaks recording and marks as Hot Clip

evilC, 2023-09-04:
Currently, when Auto Record is on, a short press of the Func button stops recording and then recording re-starts. A long press changes fan speed, which I never use.
What I would like is for a long press of the Func button to break recording as short press currently does, but also mark the previous clip as Hot

SAMUD, 2023-09-26:
Why not via - for example - double press instead of removing a feature other pilots are using?

evilC, 2023-09-26:
It could always be an option. I do not really like double-presses, too easy to get it wrong

evilC, 2023-10-27:
It looks like we may be getting context menus, which would be a much better place for this. See [here](https://github.com/hd-zero/hdzero-goggle/issues/98#issuecomment-1783604520)

evilC, 2023-12-16:
@SumolX any chance of re-picking this up now that #367 is merged into master.
Either that, or if @Master92 could pick this up, would appreciate it.