hd-zero/hdzero-goggle #344: Feature Request : Record HDMI with Audio from Mic or Line In

Yith1, 2023-09-05:
I love the Mic In/Line In Audio recording I get with the goggles when flying with HDZero.

I've been playing with the Walksnail VRX for fixed wing planes and the gyro stabilised on-board recording and the recordings are not the same... there's no audio! No chatter between the pilot and those around him. No sound as the plane whooshes past above.

Would it be possible to have the goggles be able to record the HDMI in to the SD card mixed with the Mic and/or Line In audio mixed in?

ligenxxxx, 2024-06-06:
HDZero Goggle cannot record or listen to audio from HDMI.

closed 2024-06-06 by ligenxxxx

Yith1, 2024-06-06:
@ligenxxxx 

No... please mix the audio from "line in" into the feed from hdmi in and record that.

This is so that a commentary can be recorded when using an external rx module (eg. Walksnail).