hd-zero/hdzero-goggle #356: Feature Request - Smaller FOV option

geordil, 2023-09-28:
Is there a way to scale down the image on the screens so the FOV is effectively smaller?

T1LT3DBDR, 2023-09-28:
The orqas pilot v2's did this and it was perfect. A 33° FOV is perfect for racing, less eye travel. Please, please we need this.

T1LT3DBDR, 2023-10-03:
It's not about scaling down vtx output, it's making the image smaller in the actual goggles, it would use less pixels. Thos video explains it l. It has nothing to do with camera FOV but everything to do with less eye travel in the goggles.

https://youtu.be/bEb1np8qTy0?si=zNiAlX-rBWuujz11

Master92, 2023-11-12:
This issue is a duplicate of #218

HazzaHFPV, 2023-12-12:
This would be a nice option indeed

ligenxxxx, 2024-06-06:
HDZero Goggle cannot support this.

closed 2024-06-06 by ligenxxxx