hd-zero/hdzero-goggle #357: Head Tracker CENTERING function on Func button. Feature request

MrMap, 2023-10-23:
I am so impressed by the Head Tracker implementation, the wireless solution in particular.
Recently I had the opportunity to set it up and to use it.
Though I am really missing a convenient way to center the tracker instantly, whenever desirable. The only way available today, AFAIK is entering the HT menu and choose enter the CENTER option - and in the settings I operate this is a hassle affecting the overall HT experience.

I suggest there are some factors contributing to a great HT experience:

1. Smooth and precise sensing and output generation (check - works fine)
2. Quick and easy way of HT centering at the push of a button (not there yet)
3. Quick and easy calibration (not there yet). 
Calibration may not be necessary every session, but due to temperature variations it might be.

**Suggested solution**
Allow, via goggles configuration menu, to assign the goggles Func. button to control CENTER and CALIBRATE - in stead of FAN control.
On a side note - so far, I never experienced a need to adjust the fan speed, especially not during a session. While the need of HT calibration and centering is urgent.

Todays Func button default is, short press - DVR record/stop, long press - FAN speed.
I suggest fully configurable options of, SHORT PRESS, double short press  - "DOUBLE CLICK", and LONG PRESS.

Personally I would configure DOUBLE CLICK to HT CENTER and LONG PRESS to HT CALIBRATE.
SHORT PRESS would remain record on/off.

Oh, and this need of easy centering and calibration - it´s not just me. Look at the Skyzone 04X HT for instance. Dedicated HT center / calibration button on the goggles, ready to use at any moment on ground or airborne. It´s there for a reason.

Master92, 2023-10-25:
Afaik, double pressing the right button does center the head tracker. I just changed some code around there for when you don't use it in #348

MrMap, 2023-10-25:
Wait, what?
So this was actually in the works already before my request? And pulled into next release?
Good news.

Master92, 2023-10-25:
Even more than that, this feature has already been implemented in February with #163. Did you try double clicking the right button when in video mode? It should already set the center point with any firmware released after this year's February.

MrMap, 2023-10-26:
Yay, it works exactly as you say! This is AWSOME!

Yesterday, when attempting a double click, I was in the HT menu - and then the double click had no effect (I can see the logic with that). In video mode it works perfect.

I suppose this useful feature will make it´s way into next version of the goggles manual :)

Master92, 2023-10-26:
Maybe... I don't know who updates the manual and when.

However - I still like your idea of assignable actions for all sorts of button clicks. Maybe I'll work something out when I've got the time for it 👍

Master92, 2023-11-03:
@MrMap I just updated my work on the freely assignable buttons. If you'd like, you can try it out by installing the [binary from my fork of this repo](https://github.com/Master92/hdzero-goggle/actions/runs/6745009605).

Be aware that I didn't have a chance to test it with my goggles yet, so I'd suggest you run the binary from your SD card instead of flashing it to your goggles, but based on my experience in the emulator, it should work as expected.

Just let me know if and when you tried it out! Otherwise, I might be able to try it out myself next Tuesday the earliest.

MrMap, 2023-11-03:
Wow, that's very cool!
I'd like like to try it as soon as I have opportunity.

MrMap, 2023-11-04:
@Master92
> Be aware that I didn't have a chance to test it with my goggles yet, so I'd suggest you run the binary from your SD card instead of flashing it to your goggles, but based on my experience in the emulator, it should work as expected.

As I was about to test your fork I realized that I _don´t know how to_ run a binary from the SD card - which I agree seems like a good idea instead of flashing into goggle.
Either I could do with a brief instruction... or we´ll just wait until Tuesday when you may get to test it yourself.
Cheers

Master92, 2023-11-05:
There's instructions on how to execute the application from the SD card [in the README.md](https://github.com/hd-zero/hdzero-goggle#loading-the-firmware).

I don't know, however, if you have to chmod the script file for execution. I did that anyway before testing if it works without 😆 

Either way, I'm going to test it soon myself and if that works, I'm gonna open a PR. I'd really appreciate if you'd comment there as soon as that's the case 👍

MrMap, 2023-11-05:
I tried to follow the instructions at my best, adding the three files (including developer.sd) in the SD card root.
No joy. The boot would not progress, just showing the initial HDZero splash screen with the four dots animations, and all fans running at pretty high speed. And it just went on like that...

I think I will leave the testing to you, but will be happy to follow and support your pull request. Your initiative is brilliant.

Master92, 2023-11-07:
@MrMap That doesn't sound too good 🙈

However - I just very basically tested this feature with my goggles, found a bug and fixed that. It should now work smoothly. I think it is now pretty safe to flash to your goggles and play around with it a little.

Feel free to leave a comment below the linked PR - maybe the feature will make it into 9.3.0 😄

PR #367 (2023-11-30): Make button and roller actions configurable for video mode
This adds a page in the settings where one can freely assign the actions that are executed when a button or the roller is operated in video mode.
I also added the ability to change the top fan speed and the OLED brightness level with the roller.

When changing a channel is requested, the left short button acts as the "Enter" button, regardless of which action is configured for it.

**Note:** This changes the behavior introduced with #348 where I double-assigned the double right click for either HT Center or Go Sleep! mode. However, as  [... cut]
closed 2023-11-30 by ligenxxxx