hd-zero/hdzero-goggle #359: Feature Request - When RTC mod present, place clips in date folder

evilC, 2023-10-27:
This would be useful because when viewing in the goggles, dates are not shown by the names (And is probably not really feasible given space constraints), so that if you wish to find something from a given day, it's often time consuming. The wheel is a bit laggy at best when scrolling through large amounts of clips, and this would probably help.

MrMap, 2023-10-29:
I think it might be nice with the date (and maybe time) presented with the file names, if doable.
This feature could be made user configurable.

Master92, 2024-01-23:
The file naming can be selected to represent the creation date and time since #379 has been merged. 

Placing the files in date-folders should be an easy task (that I just implemented on a private branch). Displaying the files sorted in folders in the playback page is a bit more difficult and thus time consuming. I'll try and work something out when I've got time and little by little over the next few days/weeks. I'll let you know when I've got something to try out 😉

evilC, 2024-01-23:
@Master92 are you saying that whilst they will be in folders on the SD card, that the folder structure will not be visible in the Playback menu? It will still be one big list?

Master92, 2024-01-24:
No what I'm saying is that right now (in the intermediate development state), the files get placed into folders and don't appear in the list at all. This has to be fixed so that the structure can be navigated before I'll even post a pull request 😅 Both things have to be implemented before this feature can be called usable.