hd-zero/hdzero-goggle #363: Menu highlight should change when in sub-menu

evilC, 2023-11-02:
When on the main menu, when you roll the wheel, the selected entry in the main menu is highlighted grey to indicate that the current sub-menu is selected.
When you click to enter that sub-menu, the highlight of the main menu remains grey and the sub-menu also has a highlight, which is also grey.
I feel that the main menu highlight should change in some way (Maybe turn green?) to indicate this state.

All too frequently I find myself accidentally long pressing once too many times and exiting to the FPV view, and I think this is why

pfeerick, 2023-11-04:
I agree... there needs to be some visual indication that you are in the subpage, and not interacting with the sidebar menu still. It could even be as simple as the red arrow *NOT* being shown as you scroll, and only being visible once you press on a menu item.

Master92, 2023-11-06:
I fiddled a little with the menu design and came up with the following proposal:

[image]

It changes the highlighting color to a lighter grey and removes the red arrow as soon as you enter a submenu.
What do you think about that?

pfeerick, 2023-11-06:
I tend to think  it makes sense for the arrow to be not shown until you enter the submenu/page, but I guess using it will tell if that makes sense or not. PR when? 🤣

Master92, 2023-11-06:
~~I see what you mean and thought the same. However the red arrow unfortunately doesn't have an alpha channel so when I tried to do that, there was an awkward grey-ish box around it. I think, we'd need to update the resource then but unfortunately, the original icon image is nowhere to be found 😣~~

You mean something like this?

[image]

Do you really think I should open a PR? I mean, this pretty significantly impacts the UX...

pfeerick, 2023-11-06:
Ouch, yeah, that can make things awkward!

Yes, that was the idea... I.e. the arrow reinforces that input is now in
the submenu/page. Which is what confused me right from the start when it
followed the highlight.

And sure, if you've coded something up, summit it... Others can try it and
should lead to some more discussion. :)

PR #368 (2023-11-30): Improve main menu navigation experience
When navigating the settings, the user has no feedback on if he/she is currently navigating the main menu or a sub menu except for the highlighting of elements in the submenu.

This PR changes the behavior of the navigation bar when entering and exiting a submenu element.
I.e. when navigating the main menu, no red arrow is shown. It only appears when entering a submenu. At this moment, the highlighting of the navigation bar changes, indicating a change in the level of navigation.

This feature is open to discussion, any opinion about it is highly appreci [... cut]
closed 2023-11-30 by ligenxxxx