hd-zero/hdzero-goggle #374: 9.3.0 HDMI In 60Hz not working

lukaszosinski, 2023-11-18:
After upgrading firmware to 9.3.0 hdmi input on 60Hz doesn't work. Screen turning on and off randomly or no video at all.

How to reproduce:

1. Connect gogles to yor PC/Laptop
2. Set resolution to 1080p or 720p 
3. Change frequency in the screen setting section (tested on Windows 11 with integrated Intel graphics card) from 50Hz (set bu default) to 60Hz

I believe I used 60Hz before and even if not perfect it worked. Confirmed with one person on hdzero messenger group that problem can be solved by downgrading back to 9.2.0.

ligenxxxx, 2023-11-20:
[HDZERO_GOGGLE-74-191-9.3.0-9cc8535.zip](https://github.com/hd-zero/hdzero-goggle/files/13406216/HDZERO_GOGGLE-74-191-9.3.0-9cc8535.zip)
@lukaszosinski Please test this fw.

lukaszosinski, 2023-11-21:
@ligenxxxx - still the same.

ligenxxxx, 2023-11-24:
[self_test.txt](https://github.com/hd-zero/hdzero-goggle/files/13458485/self_test.txt)
**self_test.txt** From ombi (Discord), it seems that the frame rate recognition of hdmi in is inaccurate.

lukaszosinski, 2023-11-24:
What does it mean to me? Should I try some workaround or Is it on hdzero firmware side?

ligenxxxx, 2023-11-27:
Please wait for the fix. I'm solving this problem.

lukaszosinski, 2023-11-27:
@ligenxxxx - I believe it solves the issue:

Tested the firmware with:

1080p50, 1080p60, 720p50, 720p60, 720p100 - all works.

I also have tried sim on PS5 to confirm and it works with 1080p60 set by PS5 by default. 

Great job! Thanks!

closed 2023-12-18 by ligenxxxx

djules75, 2024-06-19:
I still have the exact same issue on windows 11 with 9.3.1. The screen flickers when using 1080p@60Hz. Works fine with 1080p@50Hz. I downgraded using the above patched version ([HDZERO_GOGGLE-74-191-9.3.0-6233610.bin.zip](https://github.com/hd-zero/hdzero-goggle/files/13472735/HDZERO_GOGGLE-74-191-9.3.0-6233610.bin.zip)) and that solved the issue.

On a side note, the modes are not correctly recognized by windows 11 and I had to manually create a custom resolution in the nvidia control panel to add 720p@100Hz.

djules75, 2024-06-20:
As a follow up: with 9.3.0, I realized HD Zero VRX wasn't working (no image, only OSD from my drones). So I decided to build the firmware from the latest main branch and changed the it66021.c file as follows (since I only care about 60Hz and 100Hz modes for using the goggles with simulators on PC):

```
    if (hact == 1920 && vact == 1080) {
        ret = HDMIIN_VTMG_1080P60;
    } else if (hact == 1280 && vact == 720) {
        ret = HDMIIN_VTMG_720P100;
    }
```

That solved the issue. I can use both mode with no issues. I initially tried to revert the change from the "More relaxed FPS detection" commit, but that didn't fix the issues.

FYI: I initially also tried changing the cable and on Mac Os X, and had the same issue.

djules75, 2024-07-02:
Final update: After doing a full re-install of my NVidia drivers on my PC, the problem disappeared using the latest firmware...