hd-zero/hdzero-goggle #375: 9.3.0 RapidFire module glitching

lukaszosinski, 2023-11-20:
After updating to 9.3.0 firmware analog functionality is sometimes unusable. Video is glitching (included link to the video). Reboot sometimes helps but not always. Tested with RapidFire module.

Steps to reproduce:

A. Sometimes the issue occurs right after gogles are powered on.

B. When video is ok try to open RapidFire menu and close it few times or go to gogels menu and back or reboot gogles.

I double checked and the issue does not occur in version 9.2.0

Link to the video: https://drive.google.com/file/d/1hQ-RKagv8ZniElR980gGpiy_AEXW6yuN/view?usp=sharing

asyncnomi, 2023-11-20:
Hello :)
I think the root source of this issue is the new VA-191, you can check [Issue_#377](https://github.com/hd-zero/hdzero-goggle/issues/377)
I get the exact same behaviour as what you describe !

ligenxxxx, 2023-11-21:
@lukaszosinski What your camera video format? NTSC or PAL?

lukaszosinski, 2023-11-21:
@ligenxxxx 

1. I'm not even receiving any signal from vtx yet it's just a rapidfire menu. Even static has weird pink color artifacts. But I double checked and I have NTSC and it is alligned with my drone settings.
2. Switching PAL/NTSC doesn't help.
3. Tested firmware HDZERO_GOGGLE-74-191-9.3.0-9cc8535 - the same issue.
4. In 9.2.0 no issue.

lukaszosinski, 2023-11-30:
@ligenxxxx here is the log file.

During the test I had to switch from NTSC to PAL to reproduce the issue but as I said previously - Sometimes it is even present right after goggles start or by going to RapidFire menu.

[self_test.txt](https://github.com/hd-zero/hdzero-goggle/files/13510932/self_test.txt)

lukaszosinski, 2024-01-11:
Hello @ligenxxxx . Any progress?

JohnnyCarvi, 2024-04-05:
Hello, i Update yesterday my goggles to the latest firmware, and i have also this issue. I use speedybee Analog VRS. Would be nice if it could be fixed.

frantzbenjamin, 2024-04-16:
Having the same issue too. Pink static artifacts. using ntsc. started with firmware update to latest. using rapidfire

ligenxxxx, 2024-04-28:
https://github.com/hd-zero/hdzero-goggle/actions/runs/8866084371
Please test this firmware.

JohnnyCarvi, 2024-05-03:
Hy, I have flown totaly about 4 hours with Analog and HD Zero so far. Everything has been smooth so far. It seems that the problem has been fixed. Thank you very much.

closed 2024-05-06 by ligenxxxx