hd-zero/hdzero-goggle #377: VA-191 Causing issue on analog

asyncnomi, 2023-11-20:
The following commit: [fix hdmi in 1080p60 glitch; 1080p30 vsync tearing](https://github.com/hd-zero/hdzero-goggle/commit/cdc0f0631be6c7aee4724d863408b39849ad25e9) is introducing a bug in analog mode:
* Sometimes (completely random) the analog video will be fill by pink pixels here is a photo to illustrate: [image]
* The issue can be solve (or introduce again) by going back to the main menu and selection expansion bay again. Sometimes I need to do this 5 times to get those pink pixels away
* I can confirm that the previous commit is completely find
* I did try to incrementaly revert back changes of hardware.c and oled.c without success
* I revert back VA from 191 to 186 without changing anything else: that fixes the issue !! (but causing tearing in the main menu which is kinda expected bacause I didn't revert back anything else accordingly)
* This issue is also the root cause of [Issue_#375](https://github.com/hd-zero/hdzero-goggle/issues/375)
* During all those tests a rapidfire vrx has been used

stoneman, 2023-11-21:
I get this too except that the pink pixels are usually white or green in my case. They seem to be dependent on what’s in view. The more black in the picture, the more green artifacts I see.

I’m also using rapidfire (1.3.3 / 1.1.6) but a  friend (same rapidfire versions) is not. We are both using ntsc.

alfonsogordon, 2023-11-21:
Exactly same issue here, using True D receiver. Can fix by going to the menu and back into analog until its clear. Didnt have the issue on the previous version.

slash128v6, 2023-11-23:
I have this issue on goggle firmware 9.3. Going back to 9.2 resolved for me.

9.3
[image]

9.2:
[image]

Master92, 2023-11-26:
I just found out that I've got the same issue on an Eachine Pro58 module running Pirx firmware.

Another thing that I noticed: Before this issue comes up, switching from main menu to video seems to take longer. At least I see a black screen for a longer time than when everything works as expected. The duration of the black screen varies. Sometimes, I have to switch input modes back and forth so that the analog module gets turned off and on again.

I didn't have this issue prior to updating DM5680 and the FPGA in the course of updating the whole goggles firmware.

to-str, 2024-01-15:
I have the same problem, TBS Fusion. Works fine on 9.2, broken as described on 9.3, also tested the newest build from the actions (specifically this one: https://github.com/hd-zero/hdzero-goggle/actions/runs/7442913409), issue still persists.

If debugging is needed, I can oblige.

lukolszewski, 2024-01-26:
Same issue here with latest firmware, in my case it is a Pro58 module. I have a DVR of an entire flight with it here: [video](https://odysee.com/@RandomVids:4d/hdz_028:1) 
[image]
On my video you can see it even after the feed has ended in the static. I'm flying in NTSC mode. I wonder if others that experienced it are flying in PAL or NTSC and if PAL, is the color of the pixels different? To me this looks like it fails to detect horisontal sync of the analog signal, but IDK for sure.

It happens maybe 10% of times I power on. Power cycling usually fixes it until next power on.

lukolszewski, 2024-01-28:
@asyncnomi or anyone else, did you ever see this start happening while flying?

Yesterday I flown for an hour on analog without touching the goggles and it didn't trigger itself (once I restarted to fix it initially). But I mostly fly indoors during the winter.

If anyone has had it happen during the flight, please post a recording if possible. I'm very interested if it only happens on initialisatation, or later during the flight too. Especially if the analog signal is lost and recovered. During my hour long session yesterday I swapped many batteries, disconnecting and reconnecting the quad and it has never happened, other than initially. 

@asyncnomi mentioned it was triggered by opening vrx menu. Is this the goggle source menu? Or on the vrx itself? Can you post a recording if the latter?

to-str, 2024-01-30:
> If anyone has had it happen during the flight, please post a recording if possible. [...]

This has never happened to me so far, always was there from the beginning, or not at all. However I have switched back to 9.2 now, so my usage was limited.

JohnnyCarvi, 2024-04-05:
Hello, i Update yesterday my goggles to the latest firmware, and i have also this issue. I use speedybee Analog VRS. Would be nice if it could be fixed.

ligenxxxx, 2024-04-28:
https://github.com/hd-zero/hdzero-goggle/actions/runs/8866084371
Please test this fw.

JohnnyCarvi, 2024-04-28:
Hi Ligen

 

Thank you! I have installed your version, and my first impression is that the problem has been solved. However, I will make some flights over the next few days and switch between Analog and HD-Zero to see if it's really resolved. I will provide feedback after some flights.

 

Best regards,

Jonatan

 

https://github.com/hd-zero/hdzero-goggle/actions/runs/8866084371

JohnnyCarvi, 2024-05-03:
Hy, I have flown totaly about 4 hours with Analog and HD Zero so far. Everything has been smooth so far. It seems that the problem has been fixed. Thank you very much.

Babdef, 2024-05-04:
I've tried this firmware too, and this is a weird one;
Picture of the camera feed seems to be consistently good now, but when I enter the menu of my TBS Fusion the picture seems to be always bad now (purple static around the osd elements).

bloudman648, 2024-05-05:
@Babdef i can confirm this

Rufics, 2024-05-16:
I have the same problem, TBS Fusion.
[image]

nerdCopter, 2024-05-29:
i had this too, which PR/merge fixes this? i see #407 is still open, but this #377 is closed.
(p.s. plugging unplugging rapidfire a couple times fixed it)

ligenxxxx, 2024-05-30:
> i had this too, which PR/merge fixes this? i see #407 is still open, but this #377 and #375 is closed. (p.s. plugging unplugging rapidfire a couple times fixed it)

#407

closed 2024-06-06 by ligenxxxx