hd-zero/hdzero-goggle #378: Bug: ELRS Backpack control on Lowband

zandorsp, 2023-11-21:
To reproduce:
- Use a Expresslrs backpack flashed TX (used TX16S) ver 1.4.1 and goggles flashed with 1.4.1 too
- Select raceband and use the lua script to change the channels R1 to R8, and see work nicely, changing the channel on goggles  and vtx at same time.
- Now select lowband on goggles and try the same on channels L1 to L8.... The vtx change the channel ok, but the goggles doesn't change.

Others functions of the backpack works nicely, like the DVR control.

zandorsp, 2023-11-21:
One question: Why is necessary to have a menu with Raceband or Lowband for selection? Why not a Lowband on/off and select the channels L1-L8,R1-R8,F1,F2,F4,E1 freely ?

zandorsp, 2023-11-22:
> * Did you unlock lowband for your vtx?
> * In fact, most people use raceband, and for them lowband is redundant.

Yes, I unlocked lowband on vtx, and the vtx is working well on lowband, this affect only the Goggles and backpack

pitts-mo, 2023-11-22:
Is there a technical issue that prevents both the normal channels and low
band channels from being visible in menu at same time?

It makes more sense to have them always visibly and just change the menu
toggle to grey them out and skip over them when selecting/scanning.

zandorsp, 2023-11-23:
I think a lowband unlock on goggles option make sense... And if unlocked lowband and normal channels is selectable freely, If locked only channels R1-R8, E1, F1,F2 and F4 selectable.

ligenxxxx, 2023-11-23:
Displaying all channels (20) at the same time will make the UI look very bloated, and I don't like complicated things. And scanning all channels at the same time will be more time-consuming than before. I did this first, but the experience was not good.

pitts-mo, 2023-11-23:
Agree it sounds bloated, unneeded, and undesirable for most users.

How about we try to support all channels at once, only
display/select/search from one band at a time, default to display/search
the band current channel exists in, move band toggle from sources page and
add a hint for quick toggle/transition feature into the channels display
page.

Maybe?
This way all channels feel more accessable for users and the ability to
switch quickly will likely improve user education..
-p

Just

zandorsp, 2023-11-23:
I have a sugestion for the UI on scan now menu:
[image]
[image]
[image]

On live view all 20 channels available on dial (L1-L8,R1-R8,E1,F1,F2,F4)
and a lowband unlock option

lesharodin, 2024-05-16:
We started flying lowband more, when we have 8 pilots in the air.
Its so frustrating going to menu and change from race to low. Please make option so all freqs available in live view

zandorsp, 2024-06-08:
> We started flying lowband more, when we have 8 pilots in the air. Its so frustrating going to menu and change from race to low. Please make option so all freqs available in live view

+1