hd-zero/hdzero-goggle #381: DVR File Format Randomly Changing from MP4 to TS

slash128v6, 2023-12-04:
Running goggle firmware 9.2.0 I have seen the DVR file format randomly change from MP4 to TS. I have not figured out how to reliably reproduce the issue and it has only happened a few times. Each time I confirmed the settings were set to MP4 but the files on the SD card were TS. I had previously sent an email to support on 09/14/2023 and they said it may be a software glitch and that they would forward it to the test engineers. It just recently happened again this past weekend. I posted on the Divimath Facebook group about it and Li Gen asked me to create an issue here. Not sure what else I can add but happy to provide any further details needed.

devova, 2024-01-22:
I've just had similar behaviour.  Firmware 9.2.0

robtimko, 2024-01-23:
Happens to me all the time, randomly. I'm not able to replicate it but it happens every 30 packs or so. 9.2.0

It's seems to actually go back to mp4 on it's own too. It's frustrating because .ts isn't supported on my iOS easily.

slash128v6, 2024-01-23:
> Happens to me all the time, randomly. I'm not able to replicate it but it happens every 30 packs or so. 9.2.0
> 
> It's seems to actually go back to mp4 on it's own too. It's frustrating because .ts isn't supported on my iOS easily.

I know it's not a solution but if it helps until a fix is released - I use VLC to convert the TS files to MP4 for use on iOS devices. I also sync the TS files to my Google Drive and can open them from there with VLC player on my iPhone.

sanderpuh, 2024-05-21:
Did any of you try the new firmware, and does the issue persist?

slash128v6, 2024-05-22:
I have not yet because I thought I saw there were still some reports of issues with analog modules on the latest firmware. But I have another set of goggles I will try to test on when I get some time.

slash128v6, 2024-05-23:
Well, I was going to try it out today but when I went to grab it the last 9.3 release has in bold letters ->

"DO NOT USE THIS FW"

Crashalotfpv, 2024-06-03:
> Did any of you try the new firmware, and does the issue persist?

9.3.1 happened twice this last weekend. I noticed when back home.

ligenxxxx, 2024-06-06:
I've had some thoughts on this recently. The problem is that the MP4 files are changed to the unexpected TS format, and I guess it will only occur when the computer is turned on for the first time. I created a PR to hopefully solve this problem. https://github.com/hd-zero/hdzero-goggle/pull/421

Crashalotfpv, 2024-06-23:
So yesterday I had to replace lipo two times and guess what:
Few first ones mp4
Second lipo ts
Third was mp4 files

Single heat flying all day back2back so did not power goggles otherwise. 
Dvr triggered manually.
The only thing was my quad was running when I powered goggles second time. Set to last so going straight to the video feed on power on. It's the only thing I can think of. Need to test.

wrightwells, 2025-06-17:
Its still present in version 9.5.0 but its looks like its due to low voltage , I was using a 2s battery and when I turned on the goggles they showed a voltage of 7.25 and recorded in TS,  earlier when they were tuned on at 7.6v and down to 7.37v and it recoreded in mp4.  
I then used a 4s at 16.35v an it recorded in mp4 again.