hd-zero/hdzero-goggle #382: 9.3.0 -critical safety -fortify intent before go_sleep init from live view

pitts-mo, 2023-12-10:
Critical:
Do not allow simple double function button press to initiate go_sleep function. The init of go_sleep function from live view should require button press conditions that include presses from two or more distinct buttons.

Some suggested alternate methods to init go sleep from live view:
1 -corded long press of both the enter and function buttons?
2 -add overlay confirmation dialog that requires an additional press from enter button on double function button press? (hide overlay dialog after a couple seconds if no enter button is pressed)
3 -add menu option to configure how user initiates go_sleep from live view? (not ideal)

My instance of critical safety issue:
On 2023-12-09 I experienced an erroneous initiation of the go_sleep function while in flight. After testing for suspected software issues and skimming through the code I have not been able to find any software related mechanism to reproduce this issue. This has left me to suspect that the fan in my HDZero Goggle may have forced humidity or other debris through the goggle in such a was as to cause erroneous left button presses.

My conclusion:
I have identified the quick double function button press to initiate go_sleep function from live view as a new potential point of flight failure that should be considered a critical safety issue. (Even without identifying a conclusive reason for my instance of go_sleep initiation during flight.)

pitts-mo, 2023-12-16:
I now believe the best resolution is to add an OSD overlay requiring enter press similar to the existing "To R8?" style channel change dialog.

evilC, 2023-12-16:
IDK man, I am kinda of the opinion that Go Sleep should simply not work AT ALL while a feed is active

pitts-mo, 2023-12-17:
Yeah, I agree. But how does the goggle identify "active"? Sadly, I think user determination is the only way to identify the received signal is "active". 

On multiple instances I have found myself concerned and frustrated when my goggle suddenly reports "To **?" when I am trying to toggle the OSD and accidentally rotate the wheel.

Both the HDZero channel change and go_sleep functions are safety issues while in live view and the no_dial.txt does not appear to work as intended (disables OSD toggle too).

I would like a menu option to disable all functions with the potential to break live video while in live view (both To channel and go sleep) .

Maybe a safer solution is to use the function button as a function button and require the function button to be held for rotating the wheel and activating the To channel function or double click enter button for go sleep function.

other thoughts?
-p

evilC, 2023-12-17:
I had this bug happen to me today. I had [this PR](https://github.com/hd-zero/hdzero-goggle/pull/368) flashed.  
I had literally flashed this firmware the night before - it happened on the maiden flight of this firmware, after about a minute's flying.
I am pretty sure that I had not touched any buttons whatsoever since turning on the goggles.
I am not wholly sure what I was running before - I think it was the latest stable release (Rev 13112023)
We need to address this as a matter of urgency.
> But how does the goggle identify "active"  

The DVR starts / stops when you have a connection, so clearly it can tell

pitts-mo, 2023-12-22:
PR #367 allows user options that reduce my concern for breaking live view during flight.  :-) 
Thank you @Master92

more info:
I still had 9.3.0 release source locally and started to dig into recording of button handling in live... before things got serious I went to re-base and I was pleased to see loads of good changes in regard to button handling :-) I do still wish go_sleep to be easy to enter... I may still attempt to evaluate a go_sleep confirmation dialog overlay.

evilC, 2023-12-22:
Confirmation overlay? You mean like a yes/no in the view? That does not sound very agreeable either. Can we not just have it so that Go Sleep simply does not work at all when in live view? I don't really see any benefit to allowing it if we are currently receiving a feed from the quad

pitts-mo, 2023-12-22:
PR #367 allows you/user to configure buttons and appears to default without
go_sleep configured anywhere.

Unfortunately, I really do not see any way to safely move determination
live video is "usable" away from the user... and there racers frequently
desire to enter go_sleep while giggle is still detecting a video.... An
option to lock out go_sleep while video is detected but requiring user
determination will always be safer.

evilC, 2023-12-22:
"Always be safer" ?!?!? WTF?
You are arguing that it is safer to obstruct someone's view with a dialog, vs not obstruct someone's view.
Of course not obstructing their view is safer.

ligenxxxx, 2023-12-28:
> Yeah, I agree. But how does the goggle identify "active"? Sadly, I think user determination is the only way to identify the received signal is "active".

I can send the FC's `g_is_armed status` to goggle.

ligenxxxx, 2023-12-28:
I plan to let goggle reproduce the problem in debug mode first. I want to find the root cause of the error first.

ligenxxxx, 2024-01-06:
From Ryan:
I had a scary thing happen with latest goggle firmware last night. I was flying and then the goggle beeped, the goggle lost video, the loading screen (progress bar) showed and 3sec later, the video was back

pitts-mo, 2024-01-06:
The fact that Ryan's instance appears to have woke itself back up automatically further aligns with my theory of erroneous button press detection(s). Can we push a new release that includes #367?  Even if this does not prevent the issue it has made me feel much safer.
-p

Isolation thoughts:
-add audible key press indications for both brief and held states?

Other thoughts:
-environmental/atmospheric state? (add additional software de-bounce?)
-is ELRS able to trigger button presses?

Master92, 2024-01-06:
I'm also curious if this issue already existed before I added the sleep shortcut. Maybe nobody noticed because the only action was centering the head tracker and too few people used this feature for this issue to arise. Or maybe a change in the FPGA firmware raised this erroneous double click detection. Afaik, this part is responsible for issuing right button clicks, right?

evilC, 2024-01-06:
> erroneous double click detection

Mine happened when the sleep shortcut was set to long press right button

evilC, 2024-01-06:
I would also like to add that as far as I can see, no safeguards have currently been implemented.
1. I suggested to simply not allow sleep if live feed is active.
2. @pitts-mo suggested a confirmation if live feed is active, so that eg racers can still use it if someone started broadcasting on thier channel.  
I would guess implementing the dialog for (2) would take more work than (1)  

Assuming that is the case, then in the short term, how about this.  
Implement (1), and have it emit a beep if it decides to not shut down due to live feed.  
That way, we can fly and if we see a suspected sleep but hear no beep, we know that it is not going through that code path.  
If we hear an unexplained beep mid-flight, we know that it IS going through that code path

Longer term, we can implement (2), but by implementing (1), we are not taking away anything that racers do not already have with a stable release, we are just not giving them the shortcut sleep for now.

ligenxxxx, 2024-01-08:
@fromNansaram — 2023/12/27 17:09 (discord)
I have problems with this new Rev 13112023 fw. First, the goggles go sleep mode frequently with no button pressed. Second turn down two button sleep mode with tracking mode on, the goggles’s recording goes on and off frequently. I could not record just 2M flight recording at the race event. It record on and off like a crazy. I think right button input filter is shorten at this FW and cause noisy button press action caused these two problems.

arulrajesh, 2024-01-15:
The exact thing happened happened to me yesterday. I don't know if it is the sleep function that's gettng trigerred, feels like some kind of crash. Happened on two differnt goggles(check race video below).

Goggles screen goes blank mid flight and comes back when I scroll the wheel. Sometimes there is a beep when it goes blank, other times there is no beep.

Time stamps are in the video description when the goggles go blank. The event vrx is still picking up video no problem. Only lost a quad so far. haven't killed anyone yet.

https://www.youtube.com/watch?v=D9kypVX-QFM

I was trying to capture the crash, on the rtsp stream on my phone. 
https://youtu.be/DAt7pS9gnJg

It happened a couple of times today, but this was the only time I was able to catch in while recording the phone screen.

bloudman648, 2024-01-17:
@arulrajesh same thing here ! during flight goggles went back into source menu and beeped. same as yours, didnt always beeping. no keys pressed.

its a NO-GO!!!! these goggles are 1 year on market and all you get is a random video breakup DURING FLIGHT while goggle is resetting?

https://www.facebook.com/groups/hdzero/posts/747064013988229/?comment_id=747068557321108&reply_comment_id=747075000653797&notif_id=1705513245311524&notif_t=group_comment_mention

evilC, 2024-01-18:
@ligenxxxx  Does this repro only on fw with the sleep mode on button code, or does this repro on latest stable?  
As mentioned in the FB group, my mate has been seeing something similar on latest stable.

bloudman648, 2024-01-18:
@evilC it happens on latest, 'stable' goggle fw..

evilC, 2024-01-18:
@bloudman648 I was asking about the self_test thing that ligen was talking about.
AFAIK, it is impossible for the scenario that he is talking about to trigger sleep on latest stable, because he states `Theoretically, when an error occurs, any function of the right key may be executed`. On the latest stable, right button is unable to trigger sleep because it does not contain the button shortcuts feature.
So this leads me to believe that what he describes is potentially not the sole cause, because it appears to also be happening (Or at least something similar) on latest stable.

bloudman648, 2024-01-18:
@evilC gotcha but sometimes the goggles went also on sleep mode, so the left and right button will be pressed spontaneously altought they were not touched.

evilC, 2024-01-18:
> That's not true, 9.3.0 introduced the shortcut for entering sleep mode from live video via a double right button click.
Ah, I was unaware of this. Thanks for clearing that up.

WRT recordings, I have certainly seen in the past unexplained breaks in recordings - not something I have noticed at flight time, but upon reviewing recordings I have certainly noticed clips that start mid-flight when my hands clearly would have been on the sticks.  
I don't use HT, so I cannot comment on that.  
I have enquired with my friend if he has HT on or not. Hopefully he had it off, and this new finding explains all the scenarios.  
@bloudman648 Right, this is what ligen was describing in his last post - he has seen this happen, so we are close to squashing this bug. The big question is, is Head Tracking turned on or off on your goggles?

bloudman648, 2024-01-18:
@evilC ye, will turn in back on bud.
It happened 2 different scenarios...
Once the goggles went into sleep mode during flight and the other scenario got me back to the source menu, like u have hold the left button and gors automatically back to hdzero source ....so it seems to me that both buttons are getting triggered due to a software issue and loss of video feed DURING FLIGHT.

evilC, 2024-01-18:
I just spoke with my mate who was seeing random sleeps on latest stable, and he had HT off, so this definitely seems to be consistent with the current theory.

bloudman648, 2024-01-18:
@evilC good to hear....

ligenxxxx, 2024-01-19:
[HDZERO_GOGGLE-74-191-9.3.0-ad8fcf1.bin.zip](https://github.com/hd-zero/hdzero-goggle/files/13984519/HDZERO_GOGGLE-74-191-9.3.0-ad8fcf1.bin.zip)
Please use this firmware for testing, it is based on 9.3.0 release version, only uart_parse() is modified.

ligenxxxx, 2024-01-19:
[5680_tx.txt](https://github.com/hd-zero/hdzero-goggle/files/13987842/5680_tx.txt)
[self_test.txt](https://github.com/hd-zero/hdzero-goggle/files/13987851/self_test.txt)
Further investigation revealed that the bug was in the 5680 not properly receiving the OSD data from the RF.

ligenxxxx, 2024-01-19:
https://github.com/hd-zero/hdzero-goggle/actions/runs/7582786131
Please test this fw.

evilC, 2024-01-19:
Flashed. Am probably going to go out for a quick fly in a couple of hours, so should be able to give it a test

evilC, 2024-01-19:
> https://github.com/hd-zero/hdzero-goggle/actions/runs/7582786131 Please test this fw.

Tested for about 5 packs this afternoon and no problems.  
Go Sleep was assigned to LP of right button - just as it was when I first encountered this issue.  

Interestingly, another mate of mine has been on 9.3.0 for ages with HT off, and had no problems. The only difference between him and the other 2 of us who were seeing it was that he is on 1w, we are on 200mw. @ligenxxxx from your analysis previously, I get the impression that reproing this issue was easier with weak signal strength?

pitts-mo, 2024-01-19:
My understanding is anyone running 9.3.0 release in HDZero mode can
experience this issue but it is more likely to occur when there is weak
signal and recieved OSD data is corrupted.

evilC, 2024-01-22:
@pitts-mo - yeah, that was my interpretation of what he said too.  
@ligenxxxx Had another 2 days of flying with this firmware. Another 2 mates tried it too. So altogether we've had 30+ flights with this firmware and no sign of any issues

bloudman648, 2024-01-22:
@evilC im still scared for the last few times it happened...
the betrayal of trust needs a time lol..
so i will only fly again if it's really fixed on the new fw...

any things to say? did u had head-tracking on or off? what about elrs backpack function?

greetings:-)

evilC, 2024-01-22:
@bloudman648 On this firmware, HT on or off makes no difference, because whether the Sleep shortcut is enabled or not is no longer tied to HT being on or off, it's tied to which button press activates sleep mode. (ie what the settings are in the Input menu)   
I made sure that sleep mode was assigned to right button (I used Long Press, my mates used double-click) - ie we did everything we could to TRY and cause the bug to happen, and we couldn't get it to happen.

bloudman648, 2024-01-22:
@evilC what about elrs backpack feature for dvr recording?

bloudman648, 2024-01-22:
https://github.com/hd-zero/hdzero-goggle/commit/9b4c78b498f48e4566ed6d4d11655125fcf33f73

Interestingly...there already was a right button long press problem in 9.1.0 (june 23).
I think it never got fixed from 9.1 to 9.3?

ligenxxxx, 2024-02-01:
I'm glad there is new progress on this bug. I've uploaded the changes to #386 
When a complete osd packet from RF is received through dm5680, it will be sent to v536.
But data from RF is not entirely correct, especially when the signal is weak.
In the past, 5680 did not check the correctness of packets, resulting in sometimes packet len errors. (Exceptionally) excessively large and incorrect packet sizes would eventually cause errors when v536 received and parsed data from 5680.
Now every osd data packet from r will be checked and will be handed over to v536 for analysis after it is confirmed to be correct.

closed 2024-02-01 by ligenxxxx

bloudman648, 2024-02-04:
in the user manual its descriped, we only need the hdzero_goggle 9.x.x.bin file for updating without the other boot,os,rx,va - bin files. should we flash the fix version that way too ? just the 9.3.1 bin file ?