hd-zero/hdzero-goggle #383: OLED burnin: Change color of OSD-elements in Google Menu

stefan74xxx, 2023-12-29:
Hi,

i am already facing some OLED burnin of the BetaFlight & HDZero OSD elements in my HDZero google -good to see in the "grey" google menu background as kind of ghost pictures.

The already implemented OSD orbit function is one nice idea to reduce the effect - and it works both on the HDZero but although on BetaFlight OSD elements. Another idea was to change the typical white color of  BetaFlight fonts (stored in /resource/OSD/FC/BTFL_000.bmp) to some more dark - e.g. dark grey because then the load on the OLED is reduced and therefore the burnin problem should be less.

But the color of the HDZero OSD is fixed at white at the moment!

So feature request:  would be nice to have another line in the OSD Menu not only switch on/off OSD elements but change the overall color (e.g. with RGB-values) by the user.

Best regards,

    Stefan

ligenxxxx, 2024-02-18:
In fact, users can load their own fonts from the sd card.

stefan74xxx, 2024-02-18:
Fully agreed - i use the same way currently to reduce the burnin. But would be easier for the user just to have three sliders for R, G and B in the menu to change the color of the currently loaded font i guess...

ligenxxxx, 2024-06-06:
This is not practical. DIY fonts do not necessarily use the same black and white as the default fonts. They may use almost any color. If there are multiple colors in a font, should they all be modified according to the set RGB? I think this is not feasible. For you, you can use some color processing in PS software to get a light-colored OSD font.

closed 2024-06-06 by ligenxxxx