hd-zero/hdzero-goggle #384: Input menu does not display currently selected option(s) after boot (master post PR #367)

pitts-mo, 2024-01-01:
While testing master from SD card over the last week I have found #367 very helpful.

However the input menu appears to display default options after the goggle has been power cycled.

my test example:
Boot goggle With develop.sh HDZGOGGLE built from master on SD card.
-> Input menu
-> Roller
-> change option value from "Switch channel" (default) to "OLED brightness"
-> back (exit to main menu)

Initial verification(s):
-Exit to live view to verify roller does adjust OLED brightness.
-Return to the Input menu and verify OLED brightness is still displayed in the Roller field.

Power cycle goggle and re-verify:
-From live view verify roller does adjust OLED brightness.
-Return to the Input menu and verify the value displayed in the Roller field is incorrectly "Switch channel".

I believe some other Input fields and selections are affected but I have only verified the above.

pitts-mo, 2024-01-06:
Hi @Master92,
Just tagging you now in case you find some time coming up.
-p

Master92, 2024-01-06:
Thanks for the tag. 

Ouch, I found the issue. I made a mistake setting the appropriate roller action in the drop-down when creating the page. Luckily, it's an easy fix and only affects the roller action. I'll fix it very soon 🙈

pitts-mo, 2024-01-06:
Yes, this issue has been resolved.
Thank you.
-p

detail: 
I built #385 and ran from SD card to find the input page is properly displaying the set Roller option after power cycle.

PR #385 (2024-01-08): Fix initial roller function setting item
Fixes #384 

@pitts-mo Can you please verify that this PR fixes your issue? I did a quick code review and found this line that should cause it.
@ligenxxxx This PR should be included in the next release, or at least as soon as #367 is released as well.
closed 2024-01-06 by pitts-mo