hd-zero/hdzero-goggle #391: Emulator crashes immediately on Linux (debian bookworm)

lukolszewski, 2024-01-28:
The emulator crashes instantly on startup, but after many tries it did eventually start once (out of maybe two dozen tries). It crashes in /src/core/input_device.c:485, where SDL_WaitEvent(&event) is called in a new thread context at the same time another SDL activity is happening in the main thread. 

The call happens in a function called thread_input_device(void *ptr) which is called from input_device_init like this: pthread_create(&input_device_pid, NULL, , NULL);

Using multithreading with SDL like this will result in crashes(only one thread can use SDL at a time). Perhaps a SDL_mutex can be used to coordinate use of SDL between the threads in the emulator. 

Here is the crash output as run with gdb:
Starting program: /home/xxxxx/hdzero-goggle/build_emu/HDZGOGGLE
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff6bfe6c0 (LWP 1579019)]
[INFO][rtc.c:rtc_init:186] rtc_init has NOT detected a battery
[INFO][rtc.c:rtc_init:190] rtc_init updating both clocks via settings
[New Thread 0x7ffff63fd6c0 (LWP 1579020)]
[New Thread 0x7ffff5bfc6c0 (LWP 1579021)]
[INFO][mcp3021.c:aduc_read_voltage:60] Voltage ADC is mcp
[ERROR][mcp3021.c:mcp_read_voltage:31] mcp_read_voltage: open error.
[New Thread 0x7ffff53fb6c0 (LWP 1579022)]
[INFO][common.c:bmi2_interface_init:95] I2C Interface
[ERROR][common.c:bmi2_error_codes_print_result:165] Error [-3] : Device not found error. It occurs when the device chip id is incorrectly read
[INFO][accel_gyro.c:init_bmi270:74] init_bmi270 done.
[INFO][accel_gyro.c:enable_bmi270:88] [Pass] BMI270 enabled.
[INFO][fans.c:fans_top_setspeed:24] fans_top_setspeed: 4
[New Thread 0x7ffff7fc17c0 (LWP 1579023)]
[New Thread 0x7ffff4bfa6c0 (LWP 1579024)]

Thread 5 "HDZGOGGLE" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff53fb6c0 (LWP 1579022)]
0x00007ffff7b688d9 in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
(gdb) bt
#0  0x00007ffff7b688d9 in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#1  0x00007ffff7c16b7b in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#2  0x00007ffff7c180eb in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#3  0x00007ffff7b62383 in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#4  0x00007ffff7b626d9 in ?? () from /lib/x86_64-linux-gnu/libSDL2-2.0.so.0
#5  0x000055555556037b in thread_input_device (ptr=0x0) at /home/xxxxx/hdzero-goggle/src/core/input_device.c:485
#6  0x00007ffff7e4d044 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#7  0x00007ffff7ecd61c in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81

lukolszewski, 2024-01-28:
I submitted https://github.com/hd-zero/hdzero-goggle/pull/392 to fix this. PLease review, test, and (hopefully) merge.

ligenxxxx, 2024-06-06:
@lukolszewski Can you fix the PR #392 ?

Master92, 2024-11-08:
#392 has been merged months ago. This issue can be closed as fixed.