hd-zero/hdzero-goggle #399: Feature request: write to DVR only drone OSD, but not goggles'

qvasic, 2024-03-14:
Currently there is an option whether to write OSD to DVR. But that switches off all OSD elements: both drone's and goggles'.
I would like to have an option to choose the source of OSD to save to DVR: drone, or drone+goggles.

SumolX, 2024-05-17:
I achieve this result by dedicating a channel to manage the OSD profile 1,2,3 where 3 will only report warnings.  I quick press the dial button on the goggles and this way I have zero elements displayed.  When I arm drone it automatically starts recording and stops recording when I disarm and my video does not have any OSD elements.

ligenxxxx, 2024-06-06:
Either record all of the OSD you see, or none of it.
HDZero Goggle cannot record only half of the OSD.

closed 2024-06-06 by ligenxxxx