hd-zero/hdzero-goggle #402: DVR video random frequent micro stutter / pausing / freezing /hiccups

BeshEater, 2024-03-24:
Hi,

I've got a problem with the video on DVR. I always get frequent micro stutters/pausing. So like every 5 -10 seconds, I get a micro freeze with distortion, and then all goes normal till the next such event. I've tried 2 SD Cards (Samsung EVO plus 128 GB U3 V30, and Samsung PRO plus 128 GB U3 V30), I also tried to reduce the partition size to 32 GB, formatted through googles, formatted through Rufus, but nothing helps, it's always the same problem. I also tested the SD Cards' speed, which consistently hit ~80 MB/s write speed via SD Card Reader, also one of them worked without any issue on my action camera, so I don't think they are both faulty somehow.

This problem affects only DVR. Direct video link through googles have no such issues when I fly in real-time

Link to sample .ts file with the problem:
https://drive.google.com/file/d/1IJe1yiEgCwmjqBt2HXc_sj6QMMYdb0aL/view?usp=sharing

Firmware version - Rev_01022024 on both HDZero googles and VTX

If you need something additional from my side, like logs, or something, please ask. Thank you in advance.

hiphurraa, 2024-04-06:
I have been having the same issues. I have also tried different SD cards. The write/read speed should be ok. The stutters and freezes are very frequent.

SumolX, 2024-05-17:
Please try the following Storage -> Repair SD and then report back.

BeshEater, 2024-05-23:
@SumolX  I'll try, but how could it actually help?

SumolX, 2024-05-23:
I'm wondering if the partition is damaged.  Running the repair could potentially resolve the issue.

BeshEater, 2024-05-25:
So after a thorough check, it seems like I couldn't reproduce the original problem. Only that this time all DVRs (doesn't matter .ts or .mp4) when played on the googles are lagging nonstop like it's playing at 20 FPS. But when I play the same DVRs on the PC it looks normal, with no issue that I can see.

I've recorded video without the repaired SD card and with the repaired SD card but it seems there is no difference - https://drive.google.com/drive/folders/1N8_NXuXQGaDVVSrTosTAotVTnJ97Tz5M?usp=sharing

ligenxxxx, 2024-06-06:
@BeshEater Please contact support@divimath.com for your aftersales, this does not look like a software issue.

ligenxxxx, 2024-06-06:
Please contact support@divimath.com for your aftersales, this does not look like a software issue.

ligenxxxx, 2024-06-06:
@hiphurraa Please upload a video.

8xx8, 2024-10-14:
I have exactly the same issues. The DVR recording playback in the goggles is extremely laggy - like 15-20fps and it freezes every 10s or so. When played on PC, the framerate is ok, but the video "breaks" (image freezes and pixelates) every 10 seconds. It doesn't matter if recorded in TS or MP4 - the problem is observed with both.
I tried different sdcards (all cards work perfectly with recording 4k60 in my gopro11), tried formatting with goggles and with SDCard Formatter, tried "repair SD" in the goggles. Nothing helps.
Here's an example:
https://drive.google.com/file/d/12gGKcOFy3mncMych_t_cB5F5yT_P3-Gd/view?usp=sharing&t=47

8xx8, 2024-10-14:
[image]
[image]

ligenxxxx, 2024-12-10:
If anyone has similar issues, please contact support@divimath.com and we may be able to resolve the issue by updating the software.

closed 2024-12-10 by ligenxxxx