hd-zero/hdzero-goggle #404: Feature request - Volume control

Cascasbah, 2024-04-12:
I have noticed that there is no way to control audio in and audio out volumes on the goggles. 

The feature would be useful to record good audio, especially when using external mics, and to review footage directly on the goggles. 

Additionally, when using the goggles via HDMI, it can be useful to control audio levels.

ligenxxxx, 2024-06-06:
[image]
range 0~31

xer271, 2024-11-10:
I messed with the Line In settings a bit today. I found that the `amixer cset numid=15,iface=MIXER,name='MIC2 boost AMP gain control'` setting was the amp control for line in (default value of 4, range of [0,7]), and `amixer cset numid=1,iface=MIXER,name='AIF1 ADC timeslot 0 volume'` controlled the line in volume (default value of 160, range of [0,255].
For my mic (BY-M110), mic positioning, and voice, I had the best results with the following:
Minimum amp gain that was still enabled, as having it completely off made the snr worse, but with it higher I could make the audio clip quite easily even with the volume turned down.
`amixer cset numid=15,iface=MIXER,name='MIC2 boost AMP gain control' 1`
And then setting the mic volume as high as I could without clipping.
`amixer cset numid=1,iface=MIXER,name='AIF1 ADC timeslot 0 volume' 155`

Here are the results:
adjusting amp gain from default of 4 down to 0:
[image]
adjusting volume in increments of 5 from 160 to 140 (with gain of 1):
[image]

I think it would be quite useful if we could have a setting in the "Record Option" menu page for setting both the gain and volume for line in.

xer271, 2024-11-18:
I now have results of testing my settings at an actual race:

with default settings (amp gain = 4, adc volume = 160):
[image]

with amp gain = 1, adc volume = 155:
[image]

For reference, the spikes at the beginning of both files are me yelling "ready" to the race director with goggles already down, the background noise in both is the quads in a gym. For the first file the very short spikes on the first are from sticks hitting the endpoints and flipping switches. For the second file the longer spikes are from me talking mid race, with the very short ones being from my sticks and switches.

These settings won't be correct for everyone's mic but hopefully the results show how useful this feature would be.