hd-zero/hdzero-goggle #411: feature request: "function button" defintion choice

DarthCedius, 2024-05-16:
hello, 

request:
give the possibility to choose what the function button does on short press and long press (on my side I'd like it to swap sources on a short press)
rational:
I fly whoops a lot and I switch from analog to HDzero all the time, so I have to go in the menu to change source which is long, and a pain as the wheel selector is behind the analog bay, my hand keeps on hitting the antenna, and end up breaking the bay cover.
have a good day

Brets0150, 2024-05-25:
I am also looking to do this. When using the head tracker feature, I frequently need to re-center, and it takes a long time to do this via the menu system.

Master92, 2024-06-03:
> I am also looking to do this. When using the head tracker feature, I frequently need to re-center, and it takes a long time to do this via the menu system.

This is already implemented in #367. I don't know, why there hasn't been a release since then. Swapping sources, as @DarthCedius would be a new feature for the buttons and could possibly be implemented as a cycle through sources. Maybe I'll try that out as soon as I've got time for it.

closed 2024-06-06 by ligenxxxx

sanderpuh, 2024-06-07:
Swapping sources has not been implemented yet, see PR #423