hd-zero/hdzero-goggle #412: Formatting SD card does not cause free space to be recalculated

evilC, 2024-05-18:
Seen on: 9.3.0-09eef33 rx:74 va:191
If the SD card is full, and you reformat the SD card, the SD card still shows as being full

evilC, 2024-06-05:
Repro'd on this build: https://github.com/hd-zero/hdzero-goggle/actions/runs/9382157957
Test method:
1. FILL your SD card - on a windows system, for an SD card `F:`, in powershell:
```
cd F:
$space = $(get-WmiObject win32_logicaldisk | where-object -Property DeviceID -EQ "F:" | select -ExpandProperty FreeSpace) ; fsutil file createNew emptyFile $space
```
1. Put card in goggles - status bar shows "SD Card Full"
1. Storage menu > Format
1. Status bar still shows "SD Card Full"

Master92, 2024-11-11:
Can you please try the artifact of [this build](https://github.com/Master92/hdzero-goggle/actions/runs/11775960009)?
I don't have my goggles at hand and therefore can't try it myself at the moment.

evilC, 2024-11-13:
I just tried, but I cannot get the bloody SD card full :(
When I use the command I posted above, I get `Error:  There is not enough space on the disk.`  
Even if I format it and it's completely empty :(

PR #464 (2024-12-10): Update free SD card space after format
In attempt to fix #412, I added a call for updating the free sd card size to the formatting method. Unfortunately, @evilC was not able to reproduce and therefore confirm the fix, but I decided to give it a shot anyway.

Closes #412
closed 2024-12-10 by ligenxxxx