hd-zero/hdzero-goggle #417: DCIM DVR folder format

Crashalotfpv, 2024-06-03:
Hi there, 
I have been using import from DVRs and gopros for years. I use dongle and my iphone to edit and post quick DVR or spend some time reviewing it on multi day competitors. I used it also to argue case with jurors multiple times. It does not work with recordings from  HDZero goggle because DVR does not use DCIM format. 
Its fairy small change:
- Needs to use DCIM folder in root folder
- inside DCIM it needs 8 character folder for the device
- Files need to be 8 character long, currently until 1000 they are 7. Easy to ensure its always 8. 
Reference: 
https://en.wikipedia.org/wiki/Design_rule_for_Camera_File_system#Directory_and_file_structure

I just came back from two day competition and I was unable to review my footage on mobile like I am used to doing for years (from fatshark, powerplay, ORQA, Gopro all use DCIM).  On a headset that otherwise excels, it is unfortunate.

Crashalotfpv, 2024-06-03:
yesterday I did little test, I replicated folder structure from my go pro next to movies folder. I had to make DCIM/<8 character folder>/ I copied one 90 fps, one 60 fps dvr recording, Initially it showed import folder with no content. I realized its because file names are 7 characters long. I added "0" and I was able to import it with dongle like I used to. So it is really about folder naming and length of name for DVR recordings.

ligenxxxx, 2024-06-07:
solved.

closed 2024-06-07 by ligenxxxx