hd-zero/hdzero-goggle #432: Strange DVR on latest build if VTX is on before goggles

alfonsogordon, 2024-08-05:
i think i've narrowed this down to only when the VTX is powered up before the goggles (or before the goggles have fully booted). If i have the VTX powered on first the DVR files come out where the image is in the top left of the frame and the rest is bright green. 
[image]

alfonsogordon, 2024-08-07:
> #434 should fix this I think, unless this isn't caused by #433

Didnt seem to fix it for me

sanderpuh, 2024-08-07:
You're right, the issue you raised is from before #433 so the issue is in something else. I have the same issue, if the goggles are booted and you start recording without opening the menu first, the recording is not properly scaled

ligenxxxx, 2024-08-26:
Does this happen only when wifi is turned on?

ligenxxxx, 2024-08-26:
[image]
This is a questionable place. It may be a problem of execution order.

sanderpuh, 2024-08-26:
I have never used the wifi option on my goggles, so if the default setting for the wifi interface is `OFF`, then it also happens when wifi is turned off. 

I have been able to reproduce the behavior, my goggles are set to boot to 'last' source. If I start the recording, it does not scale the DVR. After going to the menu, and back to the source, the DVR is scaled. So perhaps if 'last' option is used for source, it skips the DVR scaling?

ligenxxxx, 2024-08-26:
@SumolX Maybe it is related to https://github.com/hd-zero/hdzero-goggle/pull/425 `page_wifi_update_settings()` is postponed but may be run in video mode.

SumolX, 2024-08-26:
@alfonsogordon and @sanderpuh could you verify with the WiFi setting enabled option set to off if you can still reproduce this issue?

chaotixtech, 2024-08-26:
I have been experiencing this issue for quite a while. Just turned wifi to OFF and rebooted the goggles, first 2 test videos are recording correctly now (eco vtx) Goggles on last commit to master and vtx 1.6.0

SumolX, 2024-08-26:
> I have been experiencing this issue for quite a while. Just turned wifi to OFF and rebooted the goggles, first 2 test videos are recording correctly now (eco vtx) Goggles on last commit to master and vtx 1.6.0

Are you running latest 9.4 firmware?  If not could you post the version reported in firmware and include the git commit id as well.

chaotixtech, 2024-08-26:
> Are you running latest 9.4 firmware? If not could you post the version reported in firmware and include the git commit id as well.
> 

75-191-9.3.0-406ad54 (file name of the download)

I assume this is 9.3.0

SumolX, 2024-08-27:
I will submit a PR which will automatically generate a new bin.  Would you mind trying it out to see if it fixes the issue.

SumolX, 2024-08-27:
> > Are you running latest 9.4 firmware? If not could you post the version reported in firmware and include the git commit id as well.
> 
> 75-191-9.3.0-406ad54 (file name of the download)
> 
> I assume this is 9.3.0

I will submit a PR which will automatically generate a new bin. Would you mind trying it out to see if it fixes the issue.

SumolX, 2024-08-27:
[HDZERO_GOGGLE-75-191-9.3.0-9f45d04.bin.zip](https://github.com/user-attachments/files/16755114/HDZERO_GOGGLE-75-191-9.3.0-9f45d04.bin.zip)

Please try this build and see if it resolves the issue.

SumolX, 2024-08-27:
@sanderpuh @chaotixtech @alfonsogordon 

Please try this build and let me know if its fixed.

[HDZERO_GOGGLE-75-191-9.3.0-9f45d04.bin.zip](https://github.com/user-attachments/files/16755121/HDZERO_GOGGLE-75-191-9.3.0-9f45d04.bin.zip)

chaotixtech, 2024-08-27:
> I will submit a PR which will automatically generate a new bin. Would you mind trying it out to see if it fixes the issue.

Flashing and testing this now.

SumolX, 2024-08-27:
@alfonsogordon @chaotixtech @sanderpuh Please try this new build.  It should now be properly fixed.
[HDZERO_GOGGLE-75-191-9.3.0-c09cd8b.bin.zip](https://github.com/user-attachments/files/16770747/HDZERO_GOGGLE-75-191-9.3.0-c09cd8b.bin.zip)

sanderpuh, 2024-08-28:
Tried, and was not able to reproduce the behavior as I've described before. No more scaling issues on my end! Note, I have only tested analog.

closed 2024-09-19 by ligenxxxx