hd-zero/hdzero-goggle #440: HDZERO googles – problem with DVR video quality

Murray-13, 2024-09-03:
After recording fine since I got them, recently my googles have not been recording video properly.  While I get good video quality on the googles when I am flying, anything that I record has very poor video quality when I play it back.

The linked image and video illustrates the poor video quality that I get from the recording on the SD card.  There are horizontal patterns on the video and the contrast seems to be too high.  I’ve also started to encounter playback problems on some of the video viewers I use.  GomPlayer often cannot even play the video (just audio), although it will usually play on windows media player.

https://www.dropbox.com/scl/fi/jo11s203yfguc055i1iyp/example-of-poor-video-from-HDZERO-vtx.png?rlkey=2wnzt0h5trsha4p4e4vyun40p&st=qqbmgwl4&dl=0

https://www.dropbox.com/scl/fi/vsqthit0w0y7dj475v8y3/hdz_002.mp4?rlkey=u0kxbn6wz63uuya7m624tbfaq&st=mj1i5g37&dl=0

The video quality that I see when I fly is fine.  But when I try to play it in the goggles, or play a copy on my computer, the quality is poor.  This happens with both analogue and HDZero vtx.

Some things that I have tried to correct the problem are:
•	recording in mp4 and TS 
•	formatting the SD card in the googles
•	doing a full format of the SD card on my computer
•	using another SD card
•	updating the googles to the latest firmware

None of these measures seem to have had any influence on the recorded video quality problem.

I’d appreciate any advice or assistance that you could give me to fix this problem.

ligenxxxx, 2024-09-03:
Please connect support@divimath.com. I can't tell if this is a software glitch, but I'm sure it's not normal.

Murray-13, 2024-09-03:
Thanks

Will do.

closed 2024-12-10 by ligenxxxx