hd-zero/hdzero-goggle #465: 9.5.0 RC1: Date-based DVR naming scheme only applies to the first file of the recording when multiple file

FractalEngineer, 2024-12-08:
Next files seem to revert to conventional.

Resets after stop/starting a new recording.

[image]

ligenxxxx, 2024-12-11:
[Uploading HDZERO_GOGGLE-75-191-9.5.0-5f776ba.zip…]()
Please test this fw.

FractalEngineer, 2024-12-22:
Thank you for your patience,

Just managed to test today and looks like that didn't do it

[image]

ligenxxxx, 2024-12-25:
I set the DVR format to MP4, the naming method to date, and set Auto scan to automatically enter hdzero after power on.
I ran multiple power on tests and found no errors.
Am I missing something that prevents me from reproducing the error?

ligenxxxx, 2024-12-25:
I use the right button to stop/start DVR in hdzero video mode. Unfortunately I can't reproduce the error you are experiencing.

FractalEngineer, 2024-12-26:
Oh ok my startup behavior is set to last,
Video file to .ts,
And perhaps most relevant this was from the analog module bay

I can enable logging for the next time if there's anything in there that could help

AndrijGorda, 2025-01-28:
I have the same issue. Auto start DVR enabled.

ligenxxxx, 2025-02-10:
I fixed it.

[HDZERO_GOGGLE-75-200-9.5.0-3b1521f.bin.zip](https://github.com/user-attachments/files/18729443/HDZERO_GOGGLE-75-200-9.5.0-3b1521f.bin.zip)

closed 2025-02-10 by ligenxxxx