hd-zero/hdzero-goggle #482: "No working" low power until armed in Betaflight

cougman, 2025-02-08:
Betaflight has this great feature (Low Power Disarm with options), why haven't we embraced this ?
This works for my analog vtx's , why not HDZero's ?
Hopefully I just missed something and need to get caught up. ....

nerdCopter, 2025-02-10:
already exists.
Mode 2 radio: ↘️↙️ = VTX-Menu
[image]

cougman, 2025-02-10:
Sorta Not the same 

image.png (view on web)

nerdCopter, 2025-02-21:
@cougman , I see your comment edited. Did it resolve your issue?

cougman, 2025-02-21:
I haven't had a chance to try it yet ,,Hopefully next week , back in the 50's  ..I'll report back asap Thanks

cougman, 2025-03-02:
: )  Had a chance to fly a few packs , It works great !  I appreciate the tip,,, somehow I missed it..

closed 2025-03-02 by cougman