hd-zero/hdzero-goggle #489: Darkened Video With Speedybeevrx

Hiten941, 2025-02-24:
hello everyone, 
                          hello guys, So I bought my hdzero goggles a month ago and i my friend bought them around a year ago. he is using tbs fusion and i am using speedybee analog vrx. for some reason my feed is darker than his i mean too dark i will share the video here. my friend's feed is the top one and the mt feed is the bottom one and also having the green bar issue

https://github.com/user-attachments/assets/a3f48f98-449c-4d11-a436-8e7f02ce0135

nerdCopter, 2025-02-24:
This is not specific to Betaflight.
I believer your camera settings need adjustments. 
  To enter camera-control, on a mode-2 radio:
 [image]

Recommend HDZero discord for HDZero support: https://discord.com/invite/kGsnEDMb2V

Also i recommend this goggle firmware as it fixes your green-screen issue: https://github.com/hd-zero/hdzero-goggle/releases/tag/v9.5.0-rc2

Hiten941, 2025-02-24:
thankyou for replying but i have already discussed this topic over discord and i guess they also dont know the reason why i am having darker screen with one type of receiver and this is only happening on hdzero goggles. when i put this vrx onm a cobra x the video just come out fantastic

vkolotov, 2025-04-23:
I have the same issue with the wifi module + rapidfire + runcam nano 2. The video is too dark. Obviously you can adjust it in the goggles but this also affects hdzero system which means you need to adjust it back for hdzero vide stream.

vkolotov, 2025-04-23:
It would be great if:
* brightness was higher for the wifi module
* or the goggles has vide settings profile so that you could switch between hdzero / analogue (probably can be automated)