hd-zero/hdzero-goggle #518: Hide signal quality level from OSD to BoxPro in AnalogMode

gumileeev, 2025-05-11:
How to hide the signal quality level from the OSD on the BoxPro in AnalogMode without disabling the OSD completely?
I try to hide the parameters VLQ, ant1, ant2, ant3, ant4 in the OSD menu, but none of this removes the signal bar in AnalogMode on the BoxPro+

[image]

pitts-mo, 2025-05-11:
It looks like it was partially prepared for OSD configuration:
https://github.com/hd-zero/hdzero-boxpro/blob/c662d5fd9fcbaf2261a5000746ff855f4e491f0a/src/core/osd.c#L300

But I do not see it added into the OSD configuration items listing:
https://github.com/hd-zero/hdzero-boxpro/blob/c662d5fd9fcbaf2261a5000746ff855f4e491f0a/src/ui/ui_osd_element_pos.c#L96

pitts-mo, 2025-05-13:
I do not have a boxpro but figured I would take another look. I several places in osd.c, ui_osd_element_pos.c and settings.h need to be touched. Is anyone looking into this?

gumileeev, 2025-05-14:
> I do not have a boxpro but figured I would take another look. Several places in osd.c, ui_osd_element_pos.c and settings.h need to be touched. I don't think I will find time for this soon... is anyone else looking into this?

I have a helmet, but I am far from programming. I took out the flash drive, but I do not see these files on it. What steps should I take to help?

pitts-mo, 2025-05-14:
hmm, I do see ligenxxxx recently made couple recent boxpro change submissions and recall Nikolas-S and SumolX worked on normal goggle OSD item position.  

@Nikolas-S / @SumolX would either of you have time or interest in adding configurable Analog RSSI OSD position for boxpro?

@gumileeev , The exposure of building/using the emulator is good learning... Suggest grab the source code from link below. Check the readme.md for instructions to build/use the emulator as well as join the discord where there are more resources to help answer questions. 
https://github.com/hd-zero/hdzero-boxpro

SumolX, 2025-05-14:
> hmm, I do see ligenxxxx recently made couple recent boxpro change submissions and recall Nikolas-S and SumolX worked on normal goggle OSD item position.
>
> [@Nikolas-S](https://github.com/Nikolas-S) / [...]

I would love too merge this codebase with the original but I do not own a boxpro currently.

gumileeev, 2025-05-18:
The day before yesterday, update 9.5.0 was released
It affected the band in analog mode: The band remained, but it is always gray, the signal level is no longer displayed.

[image]

ch1seL, 2025-05-28:
> The day before yesterday, update 9.5.0 was released It affected the band in analog mode: The band remained, but it is always gray, the signal level is no longer displayed.
> 
> [image]

same issue after updating to 9.5.0