hd-zero/hdzero-goggle #520: VTX update via boxpro

gumileeev, 2025-05-16:
I bought a BoxPro helmet. I flew on an analog video transmitter, bought a quadcopter with a VTX Eco HDZiro.
I decided to update the VTX system, and found out that the helmet does not have a connector for updating the video transmitter. Also, the system menu does not have an item for updating the VTX.
I found a photo of a disassembled helmet on the Internet and apparently there is an unsoldered connector for updating inside. Can it be used for updating? If it is suitable for updating, then the system still lacks the necessary menu item. Tell me, are there any changes planned in this direction? Or can BoxPro owners only count on friends with HDZiro glasses?

[image]
https://github.com/user-attachments/assets/5790f425-ac0b-4184-9d0c-65e921ba0a9c

benryy, 2025-08-29:
you can buy the seperate firmware flasher

ligenxxxx, 2025-09-06:
This doesn't work in hardware.

closed 2025-09-06 by ligenxxxx